homeworkc/lazycast #2: Lazycast no audio from dac and no video on screen

TheBrage, 2018-09-19:
I just set up Lazycast, i installed the requirements and then did git cloned Lazycast and chmodded every thing

homeworkc, 2018-09-19:
Please let me know what type of sources and receiver are you using. In the mean time, you can try to use player 1 (by modifying d2.py) and see if it works.

homeworkc, 2018-09-19:
Also, do you see messages like 'm3 success' or 'm6 success' in the console? Was it stuck at any particular stage?

homeworkc, 2018-09-21:
Does this solve your problem?

homeworkc, 2018-09-21:
You may also want to try lazycast on a clean installed Raspbian. lazycast should works with the unmodified wpa_supplicant that comes with most of distribution. If the wpa_supplicant is modified, it may run into some trouble controlling the wifi interface.

closed 2019-01-13 by homeworkc