homeworkc/lazycast #3: Problems faced in lazycast

superuser789, 2018-10-12:
It works fine when raspberry pi 3 is rebooted. But everytime it takes time to display pin. Before that it goes on the loop like `Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-wlan0-5'
Available interfaces:
'INTERFACES' command timed out.
Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
OK
`
After waiting for 10-15 minutes, it finally shows pin and then works flawlessly when connected. Once disconnected,  it does not work when I try to connect again and I have to reboot to make it work. 

Can following issues be solved ?
- Making process to display pin fast 
- To work again without rebooting when it is disconnected
- Some times lazycast doesn't show up even when display pin is shown

homeworkc, 2018-10-12:
1. A simple workaround right now is that if you see that lazycast is going on the loop, simply abort by ``ctrl+c`` and re-run ``all.sh``. This should work most of the time. You should only need to reboot when this trick does not work. Also, it should not take more than 1~2 mins for lazycast to get hold of the p2p interface. (That is, you should stop and re-run ``all.sh`` when it does not succeed in, say, 2 mins.) 
FYI, I added the loop specifically for Raspberry Pi 3. On other platform (e.g., Ubuntu and Intel Wifi chip), the command ``sudo wpa_cli p2p_group_add`` always succeeds. Personally I think it has something to do with p2p function in the driver for Broadcom chips. You can also change line 30 and line 35 in ``all.sh`` to see whether you can find a faster way of running ``sudo wpa_cli p2p_group_add`` successfully. (Note that on Raspberry Pi 3 sometimes the command will create an interface then die soon after. That's why we need to wait.)
2. What is the casting device you use? In the case of Windows 8.1/10 with Pi 3, if they had paired successfully, they should keeps the authentication information and the routing information. The Pi 3 also will keep the p2p interface available until the next reboot. So it should be a really fast process to cast a previously connected device. Make sure you follow the sequence of running ``all.sh`` THEN trying to connect on the casting device.
3. My guess is that this is still a pairing issue. Basically after you enter the pin, Pi 3 will act as the DHCP server which responds to ARP request (from a casting device to get an IP). Sometimes the casting device either does not send the request or does not change its IP upon receiving the response. If the casting device does not have the correct IP, lazycast will just wait forever for the first message, which should come from the casting device. I find that in these cases you should just reboot Pi 3 and try again.

superuser789, 2018-10-14:
Now, I am running on RPi zero w.  WPS pin generates in one go and it connects to source devices without any problem. 
But I want to run with built-in wifi as hotspot and  USB dongle TP-Link TL-WN722N as lazycast receiver. I have setup built-in wifi wlan0 as hotspot with following settings ::
`# cat /etc/dhcpcd.conf | grep -v "#" | sed '/^$/d'`   
`hostname
clientid
persistent
option rapid_commit
interface wlan0
nohook wpa_supplicant
static ip_address=192.168.43.1/24
static routers=192.168.43.1
`
`#  sudo nano /etc/hostapd/hostapd.conf`
`interface=wlan0
driver=nl80211
country_code=IN
ssid=MyHotspot
wpa_passphrase=somehotspotpassword
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK
hw_mode=g
channel=11
beacon_int=100
`
`# cat /etc/dnsmasq.conf | grep -v "#" | sed '/^$/d'`
`domain-needed
interface=wlan0
dhcp-range=192.168.43.2,192.168.43.10,12h
dhcp-mac=set:client_is_a_pi,B8:27:EB:*:*:*
`

The problem is when running all.sh , Pin is generated and I can see lazycast under receiver menu in devices. But after entering pin , It doesn't connect at all.

How to make it work with built-in wifi as hotspot ?

homeworkc, 2018-10-14:
Short answer: you should change all ``wpa_cli`` command in ``all.sh`` to specify the TL-WN722N interface.

superuser789, 2018-10-14:
wlan1 is the TL-WN722N interface. 
all.sh selects it automatically.
 #####################
- Selected interface 'wlan1'
- Available interfaces:
- wlan1
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'wlan1'
- OK
- Selected interface 'p2p-wlan1-0'
- Available interfaces:
- p2p-wlan1-0
- wlan1
- Selected interface 'p2p-wlan1-0'
- Available interfaces:
- p2p-wlan1-0
- wlan1
- p2p-wlan1-0
- The display is ready
- 04319658
##########################
So, lazycast  is runnning on the right interface.

homeworkc, 2018-10-14:
What is the output of ``ifconfig``?

superuser789, 2018-10-14:
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 160.204.110.103  netmask 255.255.0.0  broadcast 160.204.255.255
        inet6 fe80::1e39:47ff:fea6:22ec  prefixlen 64  scopeid 0x20<link>
        ether 1c:39:47:a6:22:ec  txqueuelen 1000  (Ethernet)
        RX packets 9027  bytes 4638606 (4.4 MiB)
        RX errors 0  dropped 1861  overruns 0  frame 0
        TX packets 2424  bytes 329332 (321.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 25  bytes 3340 (3.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 25  bytes 3340 (3.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

p2p-wlan1-0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.101.1  netmask 255.255.255.0  broadcast 192.168.101.255
        inet6 fe80::a0f3:c1ff:fe0c:c001  prefixlen 64  scopeid 0x20<link>
        ether a2:f3:c1:0c:c0:01  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 60  bytes 10499 (10.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

ppp0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1492
        inet 101.234.215.115  netmask 255.255.255.255  destination 101.234.215.1
        ppp  txqueuelen 3  (Point-to-Point Protocol)
        RX packets 4183  bytes 4375615 (4.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2304  bytes 255063 (249.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.43.1  netmask 255.255.255.0  broadcast 192.168.43.255
        inet6 fe80::ba27:ebff:fe99:d0b9  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:99:d0:b9  txqueuelen 1000  (Ethernet)
        RX packets 2630  bytes 285207 (278.5 KiB)
        RX errors 0  dropped 6  overruns 0  frame 0
        TX packets 4324  bytes 4544864 (4.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.43.5  netmask 255.255.255.0  broadcast 192.168.43.255
        inet6 fe80::a2f3:c1ff:fe0c:c001  prefixlen 64  scopeid 0x20<link>
        ether a0:f3:c1:0c:c0:01  txqueuelen 1000  (Ethernet)
        RX packets 252  bytes 33643 (32.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 47  bytes 6595 (6.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

homeworkc, 2018-10-14:
Before you enter the pin, you can actually issue ``wpa_cli -ip2p-wlan1-0`` command in another terminal to see the activity on ``p2p-wlan1-0``. Do you see something like "AP-connected"?

superuser789, 2018-10-14:
My Windows PC tried to connect and then failed to connect.
Here is the output ::
```
Interactive mode

<3>WPS-ENROLLEE-SEEN 26:8c:a0:c1:5b:bd cf0e52d4-b11f-4a70-9b95-d83b8125dbdc 1-00                                                                                                             50F200-0 0x11e8 0 0 [DESKTOP-2SFGKSP]
<3>WPS-ENROLLEE-SEEN 26:8c:a0:c1:5b:bd cf0e52d4-b11f-4a70-9b95-d83b8125dbdc 1-00                                                                                                             50F200-0 0x11e8 0 0 [DESKTOP-2SFGKSP]
<3>WPS-ENROLLEE-SEEN 26:8c:a0:c1:5b:bd cf0e52d4-b11f-4a70-9b95-d83b8125dbdc 1-00                                                                                                             50F200-0 0x11e8 0 0 [DESKTOP-2SFGKSP]
<3>CTRL-EVENT-EAP-STARTED 26:8c:a0:c1:5b:bd
<3>CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
<3>CTRL-EVENT-EAP-STARTED 26:8c:a0:c1:5b:bd
<3>CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
<3>CTRL-EVENT-EAP-PROPOSED-METHOD vendor=14122 method=254
<3>WPS-REG-SUCCESS 26:8c:a0:c1:5b:bd cf0e52d4-b11f-4a70-9b95-d83b8125dbdc
<3>WPS-SUCCESS
<3>CTRL-EVENT-EAP-FAILURE 26:8c:a0:c1:5b:bd
<3>WPS-ENROLLEE-SEEN 26:8c:a0:c1:5b:bd cf0e52d4-b11f-4a70-9b95-d83b8125dbdc 1-00                                                                                                             50F200-0 0x11e8 0 0 [DESKTOP-2SFGKSP]
<3>CTRL-EVENT-EAP-STARTED 26:8c:a0:c1:5b:bd
<3>CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
<3>WPS-ENROLLEE-SEEN 26:8c:a0:c1:5b:bd cf0e52d4-b11f-4a70-9b95-d83b8125dbdc 1-00                                                                                                             50F200-0 0x11e8 0 0 [DESKTOP-2SFGKSP]
<3>AP-STA-CONNECTED 26:8c:a0:c1:5b:bd p2p_dev_addr=16:8c:a0:c1:5b:bd
<3>AP-STA-DISCONNECTED 26:8c:a0:c1:5b:bd p2p_dev_addr=16:8c:a0:c1:5b:bd
<3>WPS-ENROLLEE-SEEN 04:c2:3e:d9:1e:75 fd80e17e-8d6b-5635-9fa4-1d8d58ffad01 10-0050F204-5 0x4288 0 0 [a32mg_htc_asia_wwe]
<3>WPS-ENROLLEE-SEEN 04:c2:3e:d9:1e:75 fd80e17e-8d6b-5635-9fa4-1d8d58ffad01 10-0050F204-5 0x4288 0 0 [a32mg_htc_asia_wwe]
>
```

homeworkc, 2018-10-14:
Actually "AP-STA-CONNECTED" indicates that p2p successfully connects. (And then Windows times out on requests and disconnects.) I think the problem now is that udhcpd may not be properly set up.

superuser789, 2018-10-14:
Lazycast was working well on rpi zero w when I had not setup hotspot.  After setting up hotspot, this problem arised.

homeworkc, 2018-10-14:
Yeah. I think multiple instants of dhcp server may cause some trouble. May be try this: https://github.com/RIOT-OS/RIOT/pull/6011?

homeworkc, 2018-10-14:
@bobogei81123 maybe you can find out if there is a solution to this.

homeworkc, 2018-10-14:
Maybe also try different DHCP servers and see if they have better interoperability. https://help.ubuntu.com/community/isc-dhcp-server

homeworkc, 2018-10-14:
On Windows, you can actually see a interface that Windows creates shortly after you enter the pin. If the IP of that interface is 192.168.43.x, it gets that from the dhcp server for hostap instead of udhcpd. Maybe you can check out [this](https://access.redhat.com/documentation/en-us/red_hat_enterprise_linux/6/html/deployment_guide/sect-configuring_a_multihomed_dhcp_server) post and simply use dhcpd for both subnets.

superuser789, 2018-10-14:
Thanks. dnsmasq was conflicting with udhcpd. So I uninstalled udhcpd and  commented 52nd line in `all.sh`.
Then added this to  `/etc/dnsmasq.conf`
 ```
interface=p2p-wlan1-0
dhcp-range=192.168.101.80,192.168.101.80,12h
```
Now , It's connecting and working.

superuser789, 2018-10-14:
I am facing some more issues like ::

1.  Randomly disconnecting after some time.
2.  Whenever, I disconnect from the source device, I need to rerun all.sh to connect again.
3.  I want to keep same Pin whenever p2p is created 
4.  Player 1 gives bad video & sound quality and Player 2 gives significant delay.

homeworkc, 2018-10-17:
1.2. This happens since the 2.4GHz spectrum is super crowded these days. (However, it may due to software bugs, too.) You might be able to force p2p to use 5.8GHz spectrum if you are not using the built-in wifi hardware.
The most important socket among all connection is the port 7236 TCP connection created at line 43 in ``d2.py``. If this connection resets or does not respond to the periodic ``keep-alive`` request from a casting device, the casting device will then tear down the connection. I suppose right now lazycast gives up too easily and there might be ways to save a connection rather than just give up.
There may exist some corner cases inside the while loop in ``d2.py``, because I do get stabler connection when the loop only focuses on responding to the ``keep-alive`` requests.
For re-connecting, only ``d2.py`` is needed for most systems. (It seems like the newest version of Windows 10 will ask you to enter pin again.) Maybe you can just add a giant loop around ``d2.py`` so that lazycast will automatically try to set up the 7236 connection again.
3. You may want to try some ``wps_pin`` commands. For now it's ``wps_pin any``. I have tried other commands before but they did not work. YMMV.
4. I think right now the a/v quality is at least on par with commercial products and the latency is better than omxplayer. The only difference (besides the fact that player 1 uses AAC while player 2 uses LPCM) between player 1 and player 2 is how they behave when packets are dropped. What surprises me is that player 2 (the code are inside the "h264" folder) feels more laggy than player 1. It is actually much simpler than player 1. It just reassembles (and re-orders) packets and sends them to the Broadcom GPU. Note that there are always trade-offs between latency and quality in real-time systems. Since the there are (and quite a lot actually) out-of-order packets, the system needs to decide whether to wait for the next packet to arrive (for quality) or to just assume the packet is lost (for latency).

codekoch, 2019-02-28:
Hi,
I'm trying to set up a hotspot at wlan1 and lazycast on wlan0.
According to superuser789 posts I added in /etc/dnsmasq.conf:
interface=<i.e. p2p-wlan0-2>
  dhcp-range=192.168.173.80,192.168.173.80,255.255.255.0,24h

in /etc/dhcpcd.conf I added:
interface <i.e. p2p-wlan0-2>
    static ip_address=192.168.173.1/24

But I wonder if the interface is available when the dhcp/dnsmaq service starts.
Therefore I use a script at runlevel 2, which contains some stuff of the lazycast all.sh script:
ain="$(sudo wpa_cli interface)"
...
p2pinterface=$(echo "${ain}" | grep "p2p-wl" | grep -v "interface")

Where p2pinterface is the interface in dnsmasq.conf and dhcpcd.conf .

ifconfig shows the following:
p2p-wlan0-2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.173.1  netmask 255.255.255.0  broadcast 192.168.173.255
        inet6 fe80::ff68:64cd:15c6:3c23  prefixlen 64  scopeid 0x20<link>
        ether ae:53:42:f6:ed:a0  txqueuelen 1000  (Ethernet)
        RX packets 16  bytes 3097 (3.0 KiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 2624  bytes 205682 (200.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

So the interface seems to be there.
And also on my AndroidPhone I see the lazycast receiver and can type in the PIN. 
But I cannot connect to the Pi. 
Everytime d2.py hangs and nothing happens. 
Seems that my phone doesn't get an IP where the d2.dy script is listening on.

Any idea?

codekoch, 2019-02-28:
Found a solution with:
sudo service dnsmasq restart

codekoch, 2019-03-02:
But one problem remains....
With the dnsmasq entry: 

interface=<i.e. p2p-wlan0-2>
dhcp-range=192.168.173.80,192.168.173.80,255.255.255.0,24h

the dhcp server works only for Android devices. With Win 10 I cannot connect. 
Only if I use the old method with udhcpd and stop the dnsmasq service:

printf "start   192.168.173.80\n">udhcpd.conf
printf "end     192.168.173.80\n">>udhcpd.conf
printf "interface       $p2pinterface\n">>udhcpd.conf
printf "option subnet 255.255.255.0\n">>udhcpd.conf
printf "option lease 60">>udhcpd.conf
sleep 3
sudo udhcpd ./udhcpd.conf

Win10 can also establish a connection. 

Very confusing, because the two dhcp entries seem to be the same....
Because I need dnsmasq for the hotspot, I'm still searching for a solution with dnsmasq....

Any help would be great!

homeworkc, 2019-03-05:
Maybe you can try using Wireshark to see the difference in traces using Windows and using Android?

codekoch, 2019-03-05:
Thanks.
But I found a working solution by clearing the file /var/lib/misc/dnsmasq.leases. 
Now only issue #21 is still a problem.
With  
sudo wpa_cli flush
and 
clearing the file /etc/wpa_supplicant/wpa_supplicant.conf at startup it still takes up to 2 minutes to find a persistent p2p-wlan interface. 
As you mentioned above sudo wpa_cli p2p_group_add creates often an interface which dies soon after....

homeworkc, 2019-03-06:
Like I mentioned before, I think this problem is specific to the Broadcom driver. Maybe some command [here](https://w1.fi/cgit/hostap/plain/wpa_supplicant/README-P2P) will help.
If you want to has faster startup time, you may want to tweak the value in https://github.com/homeworkc/lazycast/blob/aba22b445d8e1ef8752234444519a6bf637830f4/all.sh#L41
and
https://github.com/homeworkc/lazycast/blob/aba22b445d8e1ef8752234444519a6bf637830f4/all.sh#L45
I also observed that sometimes issuing 2 ``wpa_cli p2p_group_add`` simultaneously works.

TheDeepSpacer, 2019-06-04:
Non of this works for me. This are my settings.
```
sudo nano /etc/dhcpcd.conf
	interface wlan0
	static ip_address=192.198.173.1/24
sudo service dhcpcd restart

sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig
sudo nano /etc/dnsmasq.conf
	interface=wlan0
	dhcp-range=192.168.173.2,192.168.173.20,255.255.255.0,24h
	
	interface=p2p-wlan0-2
	dhcp-range=192.168.173.80,192.168.173.80,12h
	
sudo nano /etc/hostapd/hostapd.conf
	interface=wlan0
	driver=nl80211
	ssid=rasppicast
	hw_mode=g
	channel=7
	wmm_enabled=0
	macaddr_acl=0
	auth_algs=1
	ignore_broadcast_ssid=0
	wpa=2
	wpa_passphrase=adm4kit12
	wpa_key_mgmt=WPA-PSK
	wpa_pairwise=TKIP
	rsn_pairwise=CCMP
```
The acces point works fine. But when i start all.sh

Available interfaces:
p2p-dev-wlan0
wlan0
Selected Interface "p2p-dev-wlan0"
FAIL

Nothing changes when i rename the p2p or hotspot into wlan1 or wlan0

mazzasette, 2019-06-25:
Hi Guys,
I implemented lazycast into my raspberry and work perfectly. I'll set PIN manually for don't have random, is it possible?  I find into the all.sh the command wps_pin but i don't know settings.

Grazie

homeworkc, 2019-07-01:
See [this issue](https://github.com/homeworkc/lazycast/issues/18) and [this](https://w1.fi/cgit/hostap/plain/wpa_supplicant/README-WPS).

Atmoz2019, 2019-07-07:
Hi, thanks for this nice piece of software!!
I was wondering: is it also possible to run this via ethernet(cable)?
If I disable Wifi on the Raspberry Pi, and connect an ethernet-cable, then all.sh says (in a continuous loop)
wlan0
Selected Interfaces:
psp2-dev-wlan0

It doesn't come so far to show the PIN.

What can I do about that?

Thanks!!!!

homeworkc, 2019-07-08:
In the standard, there are two connection methods: WiFi p2p and TDLS. As far as I know, almost all devices only support WiFi p2p.
There is a related extension proposed by Microsoft called [Miracast over Infrastructure](https://docs.microsoft.com/zh-tw/openspecs/windows_protocols/ms-mice/). Although in this extension the discovery of supporting devices still requires WiFi p2p, it should provide much better user experience by using (reliable) TCP connection on other interfaces (e.g., Ethernet) for the actual audio/video streaming. I plan to support it eventually if some [issues](https://github.com/homeworkc/lazycast/issues/12) can be resolved. (Any help would be appreciated!)
Also, in the newer version of the standard, there is an option to switch audio/video streaming from UDP connection to TCP connection, but last time I tested it, no source has support it yet.