homeworkc/lazycast #4: can not establish connection

kiai72, 2018-11-02:
i tried this on a clean raspbian desktop on a 3b+.  it shows me the "display is ready" + the pin.
first i tried to connect from a android device but lazycast immediately crashed - so i thought it might be because of android.
So i rebooted the pi, restartet lazyboot and tried with win10. win10 finds "lazycast", asks for the pin.
But when i give the pin it takes about 2 minutes then windows says it can't connect.
there's no message in the lazyboot console window. it's still running.

any ideas what i should check ?

homeworkc, 2018-11-02:
Control messages use TCP, port 7236.
Audio/Video data use UDP, port 1028.
UIBC requires another TCP connection, which does not matter (and it's chosen by Windows anyway) if you don't use UIBC.

kiai72, 2018-11-04:
this still does not work (mcafee) - it only works when i set WUDFHost.exe to allow everything - but that's restricted for me.

homeworkc, 2018-11-05:
That's weird. Maybe it requires UDP ports 67 and 68 for DHCP?

kiai72, 2018-11-06:
does not help either. it reaches m4 success, then afterwards:
    socket.error: [Errno 32] Broken pipe
in line:
   sock.sendall('RTSP/1.0 200 OK\r\nCSeq: 4\r\n\r\n')    (d2.py)

homeworkc, 2018-11-06:
Try deleting this line: https://github.com/homeworkc/lazycast/blob/master/d2.py#L111.

kiai72, 2018-11-07:
after deleting this line i got:

OPTIONS * RTSP/1.0
Require: org.wfa.wfd1.0
CSeq: 1

RTSP/1.0 200 OK
Server: MSMiracastSource/10.00.15063.0608 guid/F61D7E93-769B-0000-780B-1EF69B76D401
Public: org.wfa.wfd1.0, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER
CSeq: 100

GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
Content-Length: 590
Content-Type: text/parameters
CSeq: 2

wfd_video_formats
wfd_audio_codecs
wfd_client_rtp_ports
wfd_display_edid
wfd_connector_type
wfd_uibc_capability
wfd2_rotation_capability
wfd2_video_formats
wfd2_audio_codecs
wfd_content_protection
wfd_idr_request_capability
intel_friendly_name
intel_sink_manufacturer_name
intel_sink_model_name
intel_sink_version
intel_sink_device_URL
microsoft_latency_management_capability
microsoft_format_change_capability
microsoft_diagnostics_capability
microsoft_cursor
microsoft_rtcp_capability
microsoft_video_formats
microsoft_max_bitrate
microsoft_color_space_conversion

RTSP/1.0 200 OK
CSeq: 2
Content-Type: text/parameters
Content-Length: 484

wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play
wfd_audio_codecs: AAC 00000001 00
wfd_video_formats: 00 00 02 04 0001FEFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none
wfd_3d_video_formats: none
wfd_coupled_sink: none
wfd_display_edid: none
wfd_connector_type: 05
wfd_uibc_capability: input_category_list=GENERIC, HIDC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=Keyboard/USB, Mouse/USB;port=none
wfd_standby_resume_capability: none
wfd_content_protection: none

m3 success

m4 success

m5 success

SETUP rtsp://192.168.101.80/wfd1.0/streamid=0 RTSP/1.0
CSeq: 101
Transport: RTP/AVP/UDP;unicast;client_port=1028

['']
[]
Traceback (most recent call last):
  File "./d2.py", line 137, in <module>
    serverport=serverport[-1]
IndexError: list index out of range

homeworkc, 2018-11-07:
How about deleting this line: https://github.com/homeworkc/lazycast/blob/master/d2.py#L75? 
Also, these articles may be useful:
https://service.mcafee.com/webcenter/portal/cp/home/articleview?articleId=TS101883
https://kc.mcafee.com/corporate/index?page=content&id=KB85497&snspd-0915
https://success.trendmicro.com/solution/1119259-firewall-setting-configuration-for-intel-widi-miracast-in-officescan

closed 2019-03-05 by homeworkc