homeworkc/lazycast #6: how to use without desktop / directFB

kiai72, 2018-11-05:
can we mod the script somehow to make it work on raspbian without desktop ? it should work on a console-only OS. For example it could call omxplayer.

homeworkc, 2018-11-05:
It should be possible. Player 1 and player 2 directly control the GPU and should be able to work on a console-only OS. What part does not work on a console-only environment?

kiai72, 2018-11-05:
i just don't see anything! player1 throws a lot of error messages. player2 does not show any error - but both do not show anything (i always see the text console).

player0: i changed "vlc" to "cvlc -Idummy" -> this at least works a bit. But looks like vlc has some permissions problems and that why it only shows "console graphics" instead of real graphics.
i'm not sure, but this might be a common problem with vlc.

i'd like to switch to omxplayer - but it does not work. OMX just says "have a nice day" - as if if could not read the stream.

tested on retropie 4.4 stretch - what's based on raspbian stretch 4.4,.

homeworkc, 2018-11-05:
The reason that omxplayer shows that message is that it does not recognize (or rather, detect) the audio stream. To solve this, select player 1 in ``d2.py`` and try to play any sound to the miracast sink while launch the omxplayer with ``omxplayer -b rtp://0.0.0.0:1028/wfd1.0/streamid=0 --live``. Alternatively, it is possible to tell omxplayer to just ignore the audio stream but I cannot remember the command off the top of my head.

homeworkc, 2018-11-06:
Stream without audio stream:
``omxplayer -b rtp://0.0.0.0:1028/wfd1.0/streamid=0 -n -1 --live``

kiai72, 2018-11-06:
hello, this works! but without audio isn't to great. So it does not have to be omxplayer if this is not possible. any other idea ?

kiai72, 2018-11-06:
i noticed something funny: i don't even need omxplayer! just use player1 - wait until it starts writing all these messages.
Then switch to another tty for example by CTRL+ALT+F2 - wait a few seconds -> video starts (with sound).
can we do this programmatically somehow ? (chvt does not work)

homeworkc, 2019-01-13:
I tried the minimal Raspbian this week and I had these findings:
The reason that the players don't work is that they need an additional package called ``libavformat-dev``. After installing this package the players should work. Note that the keyboard/mouse remote inputs do not work on Direct FB systems since the current implementation uses x window.

closed 2019-03-05 by homeworkc