homeworkc/lazycast #7: Is this project still maintained?

zvonimirfras, 2018-11-07:
Do you accept new feature requests? Are you working on making this a set-and-forget product?

How much time do you spend working on this project?

homeworkc, 2018-11-07:
What feature do you have in mind? There are newer wifi display standards that have some useful features (e.g., transmitting A/V data over TCP) but even Windows does not support those. What part makes it not set-and-forget?
I will look at the code and issues from time to time but I cannot guarantee anything.

zvonimirfras, 2018-11-08:
Right now users have to re-run `./all.sh` for every new connection and go to terminal to see what the pin code was.
A more user friendly way would be that there's a daemon that runs at boot, possibly paired with a trey tool that allows configuring. Pin would show on the screen (similar to appletv) when user initiates a connection.
This way users don't have to think about going to the terminal to see what the code was, don't have to edit source to configure and don't have to think about rerunning it when the connection drops. Trey tool can also monitor for status of the connection.

homeworkc, 2018-11-12:
At one time I was thinking about writing a GUI. I suppose that could be a direction to work on. The tricky part would be cooperation across scripts and codes. It is certainly possible to query the p2p device and generate a pin when pi sees a casting device, but the trouble of cooperation across shell script, python and c code remains. Writing all the codes in c code is certainly possible, but that would sacrifice portability.

homeworkc, 2018-12-06:
The latest commit keeps track of previous connected devices, does not require entering PIN for reconnection and is always ready for new connection when no devices are currently connected. I think it partially meets your need.

mazzasette, 2019-06-25:
Hi Guys,
I implemented lazycast into my raspberry and work perfectly. I'll set PIN manually for don't have random, is it possible? I find into the all.sh the command wps_pin but i don't know settings.

Grazie

sbaier1, 2019-10-10:
i added a systemd unit in #38 if you still need a way to auto-start this

sbaier1, 2019-10-10:
also pinging @homeworkc to review the PRs that are currently open if you get the time.

closed 2020-04-25 by homeworkc