homeworkc/lazycast #8: Don't enter pin code on win10

xiaoyu-ebox, 2018-11-11:
Hello, thank you very much for this great project.

I'm using my android phone to test that it's OK， occasionally no display on the pi desktop.
But when I use win10 as a source, win10 don't pop the window to enter pin code, but to enter it on the other end(Raspberry Pi).
[image]

homeworkc, 2018-11-12:
What is the version of Windows 10 that you're using? I have not observed such behavior on my machine.

xiaoyu-ebox, 2018-11-13:
My version of Windows is the Win10_SingleLang_Chinese(Simplified)_x64 (v1803).
[image]

CaptainMegane, 2019-05-01:
HI! I'm using Windows 10 Version 1809 Build 17763.437 and have the same issue, that I do not get prompted for the PIN to connect to device.

Even on Android 9 phone there is no such prompt to enter a pin. This software would be very nice to have and use, if it would work without issues ;-)

homeworkc, 2019-05-08:
It sounds like the problem is on the Pi side. Have you installed any other programs that might mess or interfere with the p2p settings? Can you try running it on a cleanly-installed Rasbian?

xiaoyu-ebox, 2019-05-08:
I tested it some time ago on win10 1903, and it worked fine

closed 2020-04-25 by homeworkc