homeworkc/lazycast #15: Raspbian Desktop (rpi3) not shown on android disconnection

raulmola, 2019-01-06:
First of all thanks and congratulations for this project ( i didn't manage todo make work other rpi3 Cast solutions)

I downloaded release 1.0.1 with git
I set player1 (h264 player didn't work)
With android redmi note 3 pro i succesfully connected from wireless display entering the pin. Both audio and video worked pretty well. 

But when i disconnect from android raspbian Desktop is not shown, nor can i use keyboard to show any shell. Unique solution is rebooting rpi3 (yesterday i managed booting It with ctrl+ alt+ del)

Any advice?  (Is this a bug)

homeworkc, 2019-01-06:
Short answer: Download the latest commit and ``make``.
Long answer:
Both players works by directly sending data to the GPU, so unless the player process is killed, it will not return to the desktop. Normally when you terminate from the casting source, it sends an teardown message and ``d2.py`` will gracefully kill all related processes. I haven't made releases for a long while so ``1.0.1`` is pretty old. New commits added a watchdog timer detecting whether the casting device is still transmitting (the heartbeat message) and timeout after approximately 70 sec (and then kill related processes).

homeworkc, 2019-01-07:
I figured that it may not be very convenient for you to build the project on your system, so I made a new release and you can use it directly.

raulmola, 2019-01-07:
I tried to build the project but It didn't build (some headers where not found), that's why i tried the binaries in 1.0.1. 

I'll try have a look later and give you feedback.

Thank you.

raulmola, 2019-01-07:
Now both players work watchdog is working properly killing processes returning to desktop.

Thank you.

closed 2019-03-05 by homeworkc