homeworkc/lazycast #16: IndexError: list index out of range

monpelaud, 2019-01-27:
Hi homeworkc,
Azycast is the tool I was looking for.
Many thanks for this project.
Some times the ssid of the raspberry is no longer visible and therefore no longer possible to connect.
I get this error message:
```
Traceback (most recent call last):
  File "./d2.py", line 161, in <module>
    serverport=serverport[-1]
IndexError: list index out of range
PIN:
31093781
PIN:
Failed to connect to non-global ctrl_ifname: p2p-wlan0-6  error: No such file or directory
```
If I try to stop all.sh with a Ctrl-c I get this error message:
```
^CTraceback (most recent call last):
  File "./d2.py", line 17, in <module>
    import socket
  File "/usr/lib/python2.7/socket.py", line 53, in <module>
    import _ssl
KeyboardInterrupt
PIN:
Failed to connect to non-global ctrl_ifname: p2p-wlan0-6  error: No such file or directory
```

I should restart the raspberry and it works again.

Bests regards

homeworkc, 2019-01-28:
The current implementation has an infinite loop outside the python code. Therefore, another instance of the python code will immediately run after ``ctrl+c``. Try using ``ctrl+z`` (or just closing the terminal). After this, maybe try turning off the wifi then turning it on again and running ``all.sh``.
Ultimately, the problem stems from unexpected termination of the p2p-wlan0-x interface. I suspect this has something to do with the wifi firmware (seems unlikely) or with some conflicting packages/configuration in the system. Starting from a certain release in 2018, the wifi p2p interface of Raspbian exhibits some weird behaviors. For example, the MAC address of the p2p-dev-wlan0 is randomized upon boot, which renders the p2p persistent group feature (i.e., remembering the connected devices) impossible. Also, the p2p-wlan0-x interface sometimes gets terminated unexpectedly. I am still trying to find the exact cause. For reference, I used to develop the system on [this](https://downloads.raspberrypi.org/raspbian/images/raspbian-2017-09-08/) image, and the interface is really stable and all the features are functioning properly.

monpelaud, 2019-01-28:
Ok, many thanks for your quick reply. I try to use the same image as you.
Bests regards

closed 2019-01-28 by monpelaud