homeworkc/lazycast #17: Using external USB WiFi dongle

twlawrenceko, 2019-01-31:
Hi,
I was trying to improve the connection and make the streaming more smooth by using a usb wifi dongle. It was Asus USB AC53 Nano, however, even I disable raspberry pi's internal wifi, the program keeps looping in Available interface: wlan0... And there's no pin that shows up.

I would like to ask if it's possible to use an external wifi chip to use lazycast. Thanks!

homeworkc, 2019-02-01:
1. Are you using Raspberry Pi 3B or 3B+? In the case of 3B+, I think it comes with 5.8GHz Wifi and you can try modifying this line:
https://github.com/homeworkc/lazycast/blob/aba22b445d8e1ef8752234444519a6bf637830f4/all.sh#L40
to 
``sudo wpa_cli p2p_group_add persistent$perstr freq=5``.
2. It is definitely possible to use external wifi chip, although the support of wifi p2p varies a lot from driver to driver. What is the output of ``ifconfig`` and ``sudo wpa_cli interface``?

twlawrenceko, 2019-02-01:
> * Are you using Raspberry Pi 3B or 3B+? In the case of 3B+, I think it comes with 5.8GHz Wifi and you can try modifying this line:
>   
>     
>       
>         [lazycast/all.sh](https://github.com/homeworkc/lazycast/blob/aba22b445d8e1ef8752234444519a6bf637830f4/all.sh#L40)
>       
>       
>            Line 40
>         in
>         [aba22b4](/homeworkc/lazycast/commit/aba22b445d8e1ef8752234444519a6bf637830f4)
>       
>       
>       
>       
>   
>           
>             
>              sudo wpa_cli p2p_group_add persistent$perstr 
>           
>       
>     
>   
>   
>   to
>   `sudo wpa_cli p2p_group_add persistent$perstr freq=5`.
> * It is definitely possible to use external wifi chip, although the support of wifi p2p varies a lot from driver to driver. What is the output of `ifconfig` and `sudo wpa_cli interface`?

I'm using RPI3B, not the latest 3B+ version.
This is the ifconfig for external usb wifi.
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 137.112.217.175  netmask 255.255.255.0  broadcast 137.112.217.255
        inet6 fe80::e8be:aee7:e068:990  prefixlen 64  scopeid 0x20<link>
        ether 2c:4d:54:ce:54:0e  txqueuelen 1000  (Ethernet)
        RX packets 11129  bytes 973099 (950.2 KiB)
        RX errors 0  dropped 8763  overruns 0  frame 0
        TX packets 174  bytes 28653 (27.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

This is the output for "sudo wpa_cli interface":
Selected interface 'p2p-wlan1-1'
Available interfaces:
p2p-wlan1-1
p2p-dev-wlan1
wlan1

wlan1 is the interface for internal wifi that's built in for RPI3B.

Thanks.

homeworkc, 2019-02-01:
It seems like the wifi dongle use a Realtek chip. Have you tried different drivers? https://wikidevi.com/wiki/ASUS_USB-AC53_Nano
(Just a side note, sometimes these drivers mess with wifi controls on the Raspbian, so backup everything before you proceed.)
Basically what we need is a p2p-dev-x interface. Also, sometimes you need to un-comment some options before you compile the driver to enable p2p function.

twlawrenceko, 2019-02-01:
I followed this Edimax installation process, I've also uninstalled and added "EXTRA_CFLAGS += -DRTW_P2P_GROUP_INTERFACE=1" to the first string of Makefile (https://github.com/jackfan108/rtl8822bu/blob/master/Makefile).

Here's the result from terminal:
`wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 137.112.217.160  netmask 255.255.255.0  broadcast 137.112.217.255
        inet6 fe80::ec75:e30e:c933:73ed  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:a7:0b:e1  txqueuelen 1000  (Ethernet)
        RX packets 26  bytes 2106 (2.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 38  bytes 9126 (8.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 137.112.217.229  netmask 255.255.255.0  broadcast 137.112.217.255
        inet6 fe80::e8be:aee7:e068:990  prefixlen 64  scopeid 0x20<link>
        ether 2c:4d:54:ce:54:0e  txqueuelen 1000  (Ethernet)
        RX packets 257  bytes 52960 (51.7 KiB)
        RX errors 0  dropped 159  overruns 0  frame 0
        TX packets 57  bytes 11899 (11.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

$ sudo wpa_cli interface
Selected interface 'p2p-dev-wlan1'
Available interfaces:
p2p-dev-wlan1
wlan1
`

Maybe I would need to disable internal wifi to make sure if its working.

twlawrenceko, 2019-02-01:
` $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 137.112.129.152  netmask 255.255.255.0  broadcast 137.112.129.255
        inet6 fe80::907c:ec6f:9ec4:33c2  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:f2:5e:b4  txqueuelen 1000  (Ethernet)
        RX packets 204  bytes 38583 (37.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 125  bytes 24395 (23.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 137.112.217.229  netmask 255.255.255.0  broadcast 137.112.217.255
        inet6 fe80::e8be:aee7:e068:990  prefixlen 64  scopeid 0x20<link>
        ether 2c:4d:54:ce:54:0e  txqueuelen 1000  (Ethernet)
        RX packets 171  bytes 36307 (35.4 KiB)
        RX errors 0  dropped 102  overruns 0  frame 0
        TX packets 56  bytes 12343 (12.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

$ sudo wpa_cli interface
Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
$ sudo iwlist wlan0 scan
wlan0     No scan results

`

I have disabled the internal wifi by blacklist the driver. It seems that it's working right now however I'm not able to scan any wifi. Is it because lazycast is running on p2p?

twlawrenceko, 2019-02-01:
The p2p is the one based on the usb wifi chip. However, when running lazycast it shows FAIL when selecting interface.

twlawrenceko, 2019-02-01:
`EXTRA_CFLAGS += -DCONFIG_P2P_IPS` Does not show any p2p interface.

twlawrenceko, 2019-02-07:
I would still like to ask if it's possible to get it working. P2P interface for my wifi USB dongle is working, but lazycast showed FAIL to select it. Thanks

homeworkc, 2019-02-09:
You can test whether the P2P interface is working or not by ``wpa_cli`` commands. For example, if ``p2p-dev-wlan0`` is the p2p interface, try ``wpa_cli -ip2p-dev-wlan0`` to see whether wpa_supplicant can control the USB dongle. If not, then I would suggest trying different drivers/driver configurations.

twlawrenceko, 2019-02-10:
> You can test whether the P2P interface is working or not by `wpa_cli` commands. For example, if `p2p-dev-wlan0` is the p2p interface, try `wpa_cli -ip2p-dev-wlan0` to see whether wpa_supplicant can control the USB dongle. If not, then I would suggest trying different drivers/driver configurations.

I am able to control with wpa_cli.
I am able to find my phone using p2p_find. Also able to do all kinds of p2p commands(i.e. p2p_stop)

`> p2p_connect xx:xx:xx:xx:xx:xx pbc
FAIL
`

It can also detect if the p2p device is lost. Why is it failing in this case?

Thanks

homeworkc, 2019-02-11:
Test whether this command works: ``p2p_group_add``

twlawrenceko, 2019-02-11:
> Test whether this command works: `p2p_group_add`

`> p2p_group_add
FAIL
<3>P2P-FIND-STOPPED 
<5>P2P: Failed to add group interface`

I also tested on other operation systems, it seems like it is defective. I'm calling a replacement and I'll report when the new ones return. Thanks

twlawrenceko, 2019-03-07:
Hi, I got another wifi chip but it also results in the same issue where it fails to add group interface. Is there other settings that I might need to enable?
Thanks

twlawrenceko, 2019-03-07:
I found out even I use the internal wifi chip on raspberry pi, it also gives me the FAIL sign. So is it another issue?

homeworkc, 2019-03-07:
Is it possible that [country code](http://lists.infradead.org/pipermail/hostap/2013-November/028876.html) has something to do with this?

twlawrenceko, 2019-03-07:
In order to verify if it's the country code issue, is there any way to make p2p visible in wpa_cli? The wifi chip is bought in U.S., and the phone is also bought in U.S., the RPI also has U.S. wireless county code. What I want to test on is to make my USB wifi chip p2p visible to phones, so I can at least try to connect from the phone end

homeworkc, 2019-03-07:
[Chapter 6](https://www.intel.com/content/dam/support/us/en/documents/edison/sb/edison_wifi_331438001.pdf)
Also, you may want to try explicitly set the country code
``sudo wpa_cli set country_code US``

twlawrenceko, 2019-03-07:
`pi@kol-pi3:~ $ sudo wpa_cli -i p2p-dev-wlan0 set country_code US
FAIL
pi@kol-pi3:~ $ sudo wpa_cli -i p2p-dev-wlan1 set country_code US
FAIL
pi@kol-pi3:~ $ sudo wpa_cli -i wlan0 set country_code US
FAIL
pi@kol-pi3:~ $ sudo wpa_cli -i wlan1 set country_code US
FAIL`

All of them failed to set country code by the method you provided.

Thanks.

homeworkc, 2019-03-08:
In that case you need to add ``country_code=US`` to ``/etc/wpa_supplicant/wpa_supplicant.conf`` and reboot.

iamogopogo, 2019-10-22:
Firstly, thanks for sharing this. I've been looking for something to cast an Oculus Quest to the Pi, which is about to get Miracast support. Until now, we could only use scrcpy to cast to a pc or Pi, but without audio. So, this will be perfect...if I can get it to work. :P

Using the RPi4's onboard chip, I'm able to get p2p-device-wlan0 recognized, but I'm mounting my Pi, and both an SSD for the OS, and a 4TB WD drive for a NAS, in a gutted out psu box/metal case. I still get wifi signal through the vent slots in the case, but I'd prefer using an external dongle for best signal strength. I'm working with an Archer T4UH TP-Link (rtl8812au), and out of the box, it only shows P2P-client and P2P-GO, with supported commands 'new_interface' and 'set_interface', if that can be used in any way.

Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO

Supported commands:
		 * new_interface
		 * set_interface

Using your tips on enabling the extra cflags, I added both the 'EXTRA_CFLAGS += -DCONFIG_P2P_IPS' and 'EXTRA_CFLAGS += -DRTW_P2P_GROUP_INTERFACE=1' to the ARM_RPI section in the Makefile, recompiled, but it's still not showing P2P-device:

```
ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT
EXTRA_CFLAGS += -DCONFIG_P2P_IPS
EXTRA_CFLAGS += -DRTW_P2P_GROUP_INTERFACE=1
ARCH ?= arm
CROSS_COMPILE ?=
KVER ?= $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
INSTALL_PREFIX :=
endif
```
```
sudo wpa_cli interface
Selected interface 'wlan0'
Available interfaces:
wlan0
```

This is the source I'm compiling:
https://github.com/abperiasamy/rtl8812AU_8821AU_linux

It works for everything else, even getting 866Mbps as an access point. It just won't configure as a p2p-device. Any ideas?

iamogopogo, 2019-10-22:
Actually, never mind. By the sounds of it, it might be the chipset, not the driver. I came across a Microsoft doc page that had a link to usb dongles guaranteed to work with Wifi Direct, and some of them are supported for Arm32, x64 and x86. I just ordered the TP-LINK TL-WN725N Wireless N Nano, for $20 CAD off Amazon. It's only 150 Mbps, but the Pi's onboard chip at 72 Mbps, is more than fast enough to stream 4K 180 VR videos, so even if that 150 is only 72 in the real world, it'll still be fine. I'll keep the 5GHz T4UH for the wireless hotspot and NAS though, I'll just have to figure out how to have them both at the same time.

https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/setupwifidirect
https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/hardwarecompatlist
https://www.amazon.ca/gp/product/B008IFXQFU

twlawrenceko, 2019-10-22:
One thing to mention about here is, even though my ASUS-AC53-Nano supports wifi direct in "Windows" it does not mean it is supported in Raspbian. I've sent out email to the manufacture and they said it is not supported in Linux, so that is the reason why it is not working.
A good tip of testing it would be disabling the on board wifi and use only from the external USB wifi.

Hope you find a good result!

iamogopogo, 2019-10-22:
Thanks. Disabling the onboard wifi and bluetooth in the /boot/config.txt was the first thing I did when trying an external dongle. But that list by Microsoft showing dongles supported for the Pi was a lucky find. At least I know the dongle's supported and will work before actually ordering it. Cheers.

twlawrenceko, 2019-10-22:
> Thanks. Disabling the onboard wifi and bluetooth in the /boot/config.txt was the first thing I did when trying an external dongle. But that list by Microsoft showing dongles supported for the Pi was [...]

Nice, is this the dongle you are talking about: TP-LINK TL-WN725N? I'm glad that it shows it supports somewhere on the internet, don't forget to tell us the results you got!

iamogopogo, 2019-10-23:
Yep...it's the 4th in the list.

`TP-LINK TL_WN725N | ARM32, x64, x86 | TP-LINK TL-WN725N Wireless N Nano USB Adapter 150 Mbps (USB/VID_0BDA&PID_8179)`

They also had the official Raspberry Pi dongle at the top of the list, but it was discontinued January 2018, and nobody seems to be selling it any more. That would have been my first choice if I could find it.

iamogopogo, 2019-10-23:
I also just came across Realtek Rtl8192Cu, which has an Amazon review with the actual iw list output. I'm thinking of cancelling my order and getting this instead.

```
Supported Ciphers:
* WEP40 (00-0f-ac:1)
* WEP104 (00-0f-ac:5)
* TKIP (00-0f-ac:2)
* CCMP (00-0f-ac:4)
* 00-0f-ac:10
* GCMP (00-0f-ac:8)
* 00-0f-ac:9
* CMAC (00-0f-ac:6)
* 00-0f-ac:13
* 00-0f-ac:11
* 00-0f-ac:12
Available Antennas: TX 0 RX 0
Supported interface modes:
* IBSS
* managed
* AP
* AP/VLAN
* monitor
* P2P-client
* P2P-GO
Supported commands:
* start_ap
software interface modes (can always be added):
* AP/VLAN
* monitor
```

https://www.amazon.com/BLUECELL-300Mbps-Realtek-Rtl8192Cu-Wireless/product-reviews/B00GYQXDQW

twlawrenceko, 2019-10-23:
> I also just came across Realtek Rtl8192Cu, which has an Amazon review with the actual iw list output. I'm thinking of cancelling my order and getting this instead.
>
> ```
> Supported Ciphers:
> * WEP40 [...]

Make sure you understand that these are the functionality that supports on "Raspbian", not other OS. Because that was the catch on my AC53Nano. Tell us how it goes after it arrives!

iamogopogo, 2019-10-23:
Well, I feel a bit foolish now...I could have sworn it said P2P-device in that list. As for it being Raspbian, that's what I'm running, so that wouldn't have been a problem... if it was the right device. I guess I'm not cancelling my order. That sucks...this had 2x the speed, and an actual antenna. :P It also would have taken 4 weeks to arrive...the TP-Link TL-WN725N will be here by Saturday. I'll let you know how it works.

iamogopogo, 2019-10-23:
There's a pair of videos (pt1 and pt2) on Youtube showing the device working.

https://www.youtube.com/watch?v=6GPv8TfZqe4
https://www.youtube.com/watch?v=-HCrfkKeIM0

twlawrenceko, 2019-10-23:
> There's a pair of videos (pt1 and pt2) on Youtube showing the device working.
> 
> https://www.youtube.com/watch?v=6GPv8TfZqe4
> https://www.youtube.com/watch?v=-HCrfkKeIM0

This seems to be very promising, I now remember my purpose of using wifi direct on external usb is wanting to improve the lagging with miracast. Mine was AC750(AC53Nano), I believe, however, I am very interested as you said this external USB WiFi might be already enough for Miracast. Tell us how the external USB WiFi Miracast goes!

iamogopogo, 2019-10-27:
Well, this is frustrating as all hell. The video shows it working, and various googles spit out threads saying the TL-WN725N (8188eu) works with WiFi Direct, but unless it's a driver issue, none of them will display '* P2P-device' under supported interface modes. I did come across a thread from 2014, where the guy was successful, but the driver linked is no longer being hosted on dropbox. It did have a link to something from an rtl8192cu driver though, that had a folder WiFi_Direct_User_Interface, where once the files inside are installed, you then just do './P2P_UI wlan0' from the directory, and it gives you a terminal with options to enable the device. But...mine's saying:

'Interface doesn't accept private ioctl...
p2p_set (8BF0): Operation not permitted

So I'm not sure what the hell(?). Unless it's a difference in version numbers of the dongle itself...my dongle's a TL-WN725N v3, and the post from 2014 might be v2 or even v1. Either that, or it is indeed the driver.

https://www.raspberrypi.org/forums/viewtopic.php?t=91891
https://github.com/ninjablocks/rtl8192cu/tree/master/WiFi_Direct_User_Interface

twlawrenceko, 2019-10-28:
> p2p_set (8BF0): Operation not permitted

You might need to setup the mode for the external wifi for p2p usage.
[https://unix.stackexchange.com/questions/402885/iwconfig-operation-not-permitted](url)

iamogopogo, 2019-10-28:
So it just needs to be ad-hoc? I figured it's not showing the 'P2P-device' under 'supported interfaces' with 'iw list', as either it's not built into the driver, or my chipset's saying it doesn't even have that feature to enable, regardless of the driver. I've even see the 'EXTRA_CFLAGS' lines suggested above commented out for various platforms in the sources, just nothing under the 'ARM_RPI' or 'BCM2709' sections, and even though I added the lines, it still doesn't show 'P2P-device', so lazycast doesn't see 'p2p-device-wlan0'. I've since realized just having P2P-GO and P2P-Client support is still considered WiFi Direct enabled, just not the way lazycast needs it. I guess with those modes you already know the IP of the device and connect direct(?). If I could stream to the Pi using P2P-Client, then that's all I'd need. If that's even possible if it doesn't report as a p2p-device.