homeworkc/lazycast #22: Reliable way of streaming in KODI

michalbielecki, 2019-03-07:
Hi.
First of all, thank you for your great work. This project is exactly what i'm looking for with expanding my home media center functionalities. Have you considered integrating your solution into XBIAN or any other distro? I'm having trouble using this with XBIAN. I managed to stream my screen using player=1, player=2 and player=3(omxplayer). The only difficulty that arises is that I need to have some player already opened and then lazycast somehow hijacks the screen and mirroring works. It works really great and I'm wondering if I could make this solution into something more generic.

homeworkc, 2019-03-08:
Do you mean that you have difficulty launching it or you don't want full-screen players?

michalbielecki, 2019-03-08:
I mean that this is inconvenient because i have to launch some other media to get player opened. It works as workaround rather than generic solution. Also, if I do open some media to get the player opened and fast forward to point where screen is not black, my stream content is mixed with whatever is on the screen(like overlaying images with transparency set).