homeworkc/lazycast #27: compilation troubles

tomakkermans, 2019-05-13:
hi, i very much need this kind of software. Reading about Lazycast i thought, well let's try it. But in the documentation, nowhere is mentioned that one need to first compile the software, of course since there are no binaries delivered but only source code. So, before doing anything else, i first type "make" and hope for a succesfull compilation.. but no; i have copypasted the output, including many compilation warnings and errors, below. Any idea why this happens and how i can solve it?
Btw: if i compile it like this, including the warnings, i can't get lazycast working properly; the connection stays for few seconds (i can see the android screen and audio on tv) but then it falls away and connection is broken;

here the compilation output:

pi@raspberrypi:~/lazycast-master $ make
make -C player/.
make[1]: Map '/home/pi/lazycast-master/player' wordt binnengegaan
cc -DOMX_SKIP64BIT -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont   -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont -g -c player.c -o player.o -Wno-deprecated-declarations
player.c: In function ‘copy_into_buffer_and_empty’:
player.c:221:6: warning: unused variable ‘buff_size’ [-Wunused-variable]
  int buff_size = buff_header->nAllocLen;
      ^~~~~~~~~
player.c: In function ‘setup_demuxer’:
player.c:470:50: warning: comparison of constant ‘0’ with boolean expression is always false [-Wbool-compare]
   if (!avcodec_open2(codec_context, codec, NULL) < 0)
                                                  ^
player.c:470:50: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
player.c:491:50: warning: comparison of constant ‘0’ with boolean expression is always false [-Wbool-compare]
   if (!avcodec_open2(codec_context, codec, NULL) < 0)
                                                  ^
player.c:491:50: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
player.c: In function ‘setup_clockComponent’:
player.c:602:23: warning: unused variable ‘clock’ [-Wunused-variable]
     OMX_COMPONENTTYPE*clock = ilclient_get_handle(*clockComponent);
                       ^~~~~
player.c: In function ‘audioplay_get_latency’:
player.c:729:16: warning: variable ‘error’ set but not used [-Wunused-but-set-variable]
  OMX_ERRORTYPE error;
                ^~~~~
player.c: In function ‘receivepkt’:
player.c:823:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
player.c: In function ‘addnullpacket’:
player.c:908:16: warning: variable ‘padpacket’ set but not used [-Wunused-but-set-variable]
  unsigned char padpacket[2048];
                ^~~~~~~~~
player.c:904:6: warning: unused variable ‘err’ [-Wunused-variable]
  int err = 1000;
      ^~~
player.c: In function ‘main’:
player.c:1318:37: warning: passing argument 3 of ‘pthread_create’ from incompatible pointer type [-Wincompatible-pointer-types]
   if (pthread_create(&thread, NULL, receivepkt, oldnodecopy) != 0)
                                     ^~~~~~~~~~
In file included from player.c:6:0:
/usr/include/pthread.h:233:12: note: expected ‘void * (*)(void *)’ but argument is of type ‘void * (*)(Nodetype *) {aka void * (*)(struct Node *)}’
 extern int pthread_create (pthread_t *__restrict __newthread,
            ^~~~~~~~~~~~~~
player.c: In function ‘copy_into_buffer_and_empty’:
player.c:169:10: warning: ‘r’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   return r;
          ^
cc -o player.bin -Wl,--whole-archive player.o -lilclient -lavformat -lavcodec -lavutil  -L/opt/vc/lib/ -lbrcmGLESv2 -lbrcmEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lpthread -lrt -lm -L/opt/vc/src/hello_pi/libs/ilclient -L/opt/vc/src/hello_pi/libs/vgfont -Wl,--no-whole-archive -rdynamic
rm player.o
make[1]: Map '/home/pi/lazycast-master/player' wordt verlaten
make -C h264/.
make[1]: Map '/home/pi/lazycast-master/h264' wordt binnengegaan
cc -DOMX_SKIP64BIT -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont   -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont -g -c h264.c -o h264.o -Wno-deprecated-declarations
h264.c: In function ‘video_decode_test’:
h264.c:413:31: warning: passing argument 2 of ‘audioplay_create’ from incompatible pointer type [-Wincompatible-pointer-types]
  if (audioplay_create(client, &audio_render, list, 4) != 0)
                               ^
In file included from h264.c:43:0:
audio.h:1:9: note: expected ‘COMPONENT_T * {aka struct _COMPONENT_T *}’ but argument is of type ‘COMPONENT_T ** {aka struct _COMPONENT_T **}’
 int32_t audioplay_create(ILCLIENT_T *, COMPONENT_T *, COMPONENT_T **, int);
         ^~~~~~~~~~~~~~~~
h264.c:494:10: warning: unused variable ‘startindicator’ [-Wunused-variable]
      int startindicator = buffer[1] & 0x40;
          ^~~~~~~~~~~~~~
[... 33 lines cut ...]
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc -DOMX_SKIP64BIT -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont   -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont -g -c audio.c -o audio.o -Wno-deprecated-declarations
audio.c: In function ‘audioplay_create’:
audio.c:55:10: warning: unused variable ‘s’ [-Wunused-variable]
  int32_t s;
          ^
At top level:
[... cut ...]
cc -O2 -Wall  -g -c controlhidc.c -o controlhidc.o 
cc -o controlhidc.bin controlhidc.o -L /usr/X11R6/lib -lX11 -lm
cc -O2 -Wall  -g -c control.c -o control.o 
control.c: In function ‘main’:
control.c:198:13: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 2 has type ‘KeySym {aka long unsigned int}’ [-Wformat=]
    printf("%x\n", *keysym);
             ^
cc -o control.bin control.o -L /usr/X11R6/lib -lX11 -lm
rm control.o controlhidc.o
make[1]: Map '/home/pi/lazycast-master/control' wordt verlaten

homeworkc, 2019-05-14:
If you just want to use lazycast, follow the instructions in the [Preparation](https://github.com/homeworkc/lazycast#preparation) section. (Also remember to install udhcpd). Pre-compiled binaries can be downloaded from the [release](https://github.com/homeworkc/lazycast/releases) page. After you ``chmod`` every binary, you should be able to use it by executing ``all.sh``.

tomakkermans, 2019-05-14:
thanks, i didnt know about the precompiled binaries. But the other problem remains identical, ill open a different issue for that

closed 2019-05-21 by homeworkc