homeworkc/lazycast #28: connection broken after two seconds

tomakkermans, 2019-05-14:
hi, first of all, thank you very much for the work.
It would be great if it worked!µ

So after setting the permissions and installing the dependencies, i run the ./all.sh script.
First it asks for the pin code, and i can connect my android phone. For a brief moment, i then see the android screen and hear the audio on the RaspberryPI , but this quickly disappears and a new pin code is asked. Here the output from the terminal:

pi@raspberrypi:~ $ cd lazycast/
pi@raspberrypi:~/lazycast $ ./all.sh
Selected interface 'p2p-wlan0-1'
Available interfaces:
p2p-wlan0-1
p2p-dev-wlan0
wlan0
already on
p2p-wlan0-1
The display is ready
PIN:
90554971
---M1--->
OPTIONS * RTSP/1.0
CSeq: 0
Require: org.wfa.wfd1.0

<--------
RTSP/1.0 200 OK
CSeq: 1
\Public: org.wfa.wfd1.0, SET_PARAMETER, GET_PARAMETER

<---M2---
OPTIONS * RTSP/1.0
CSeq: 100
Require: org.wfa.wfd1.0

-------->
RTSP/1.0 200 OK
CSeq: 100
Public: org.wfa.wfd1.0, GET_PARAMETER, SET_PARAMETER, SETUP, PLAY, PAUSE, TEARDOWN

---M3--->
GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 1
Content-Length: 127
Content-Type: text/parameters

wfd_audio_codecs
wfd_video_formats
wfd_coupled_sink
wfd_client_rtp_ports
wfd_standby_resume_capability
wd_initial_buffer

<--------
RTSP/1.0 200 OK
CSeq: 2
Content-Type: text/parameters
Content-Length: 485

wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play
wfd_audio_codecs: LPCM 00000002 00
wfd_video_formats: 00 00 02 04 0001FEFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none
wfd_3d_video_formats: none
wfd_coupled_sink: none
wfd_display_edid: none
wfd_connector_type: 05
wfd_uibc_capability: input_category_list=GENERIC, HIDC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=Keyboard/USB, Mouse/USB;port=none
wfd_standby_resume_capability: none
wfd_content_protection: none

---M4--->
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 2
Session: 00000000
Content-Length: 249
Content-Type: text/parameters

wfd_audio_codecs: LPCM 00000002 00
wfd_video_formats: 00 00 02 04 00000080 00000000 00000000 00 0000 0000 00 none none
wfd_presentation_URL: rtsp://192.168.173.80/wfd1.0/streamid=0 none
wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play

<--------
RTSP/1.0 200 OK
CSeq: 3

---M5--->
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 3
Session: 00000000
Content-Length: 27
Content-Type: text/parameters

wfd_trigger_method: SETUP

<--------
RTSP/1.0 200 OK
CSeq: 4

<---M6---
SETUP rtsp://192.168.101.80/wfd1.0/streamid=0 RTSP/1.0
CSeq: 101
Transport: RTP/AVP/UDP;unicast;client_port=1028

-------->
RTSP/1.0 200 OK
CSeq: 101
Date: Tue, May 14 2019 06:15:39 GMT
Session: 00000000
Transport: RTP/AVP/UDP;unicast;client_port=1028;server_port=19022

['RTSP/1.0 200 OK\r\nCSeq: 101\r\nDate: Tue, May 14 2019 06:15:39 GMT\r\nSession: 00000000\r\nTransport: RTP/AVP/UDP', 'unicast', 'client_port=1028', 'server_port=19022\r\n\r\n']
['server_port=19022\r\n\r\n']
19022
<---M7---
PLAY rtsp://192.168.101.80/wfd1.0/streamid=0 RTSP/1.0
CSeq: 102
Session: 00000000

-------->
RTSP/1.0 200 OK
CSeq: 102
Date: Tue, May 14 2019 06:15:39 GMT

---- Negotiation successful ----
idrport:42799
audiodest:0
oldcc 0 cc 4

PIN:
85127845
PIN:
42664680
PIN:
13581770
PIN:
49752687

tomakkermans, 2019-05-20:
Nobody has this issue?! Its driving me nuts! :)

homeworkc, 2019-05-21:
I would suggest testing different sources (like a Windows computer) and players (e.g., player_select = 1). Also try disable mouse/keyboard input (enable_mouse_keyboard = 0). From the log, the behavior is a bit weird. 
These lines:
``
idrport:42799
audiodest:0
oldcc 0 cc 4
``
are the outputs of the player (i.e., the binary files). In a normal situation, the python script (``d2.py``) should print out more messages (like keep-alive messages, or, at the very least, a teardown message) after this. My guess is that the wifi is very crowded or the wifi card isn't working properly. (You use the built-in wifi, right?) You may also try modifying this line:
https://github.com/homeworkc/lazycast/blob/db425d30a868b752c16e9e3dcd5ec3439a5c6db8/all.sh#L40
to 
``sudo wpa_cli p2p_group_add persistent$perstr freq=<freq in MHz>``
where you can set the frequency to a different [wifi channel](https://en.wikipedia.org/wiki/List_of_WLAN_channels) in 2.4GHz or 5GHz bands. You may also try ``freq=2`` or ``freq=5`` to see whether the auto-selection does a better job.

tomakkermans, 2019-05-21:
ok', i'll try it asap (this evening); Is the program compatiable with wifi dongles? or written specially (meant for) the built-in wifi?

homeworkc, 2019-05-21:
It really depends on what chip the dongle uses. You can use [WikiDevi](https://wikidevi.com/wiki/Main_Page) to look up the IC part number. It should work without problems for some chips, but it may not work so smoothly for others. I only tested it on built-in wifi (3B and 3B+) chips.

tomakkermans, 2019-05-21:
sadly.. no improvement. Changing the channel even caused my android phone to not detect any lazycast device in the list. But thanks for your suggestion. I'll try lazycast again in some months, hopefully with some better success! cheers