homeworkc/lazycast #31: Pi as Hotspot and lazycast issue

TheDeepSpacer, 2019-05-23:
I´ve tried to setup my Pi3 to a miracast receiver wich worked perfectly.
As soon as I startet to turn my Pi into a standalone network
(https://www.raspberrypi.org/documentation/configuration/wireless/access-point.md)(so its not connectet to my wifi an can receive the miracast) lazycast wont work anymore.
Tried everything: just one dhcp, solutions from other issues. Nothing worked for me.
Both parts, lazycast and the accespoint both work separately.

homeworkc, 2019-05-23:
I think you already figure out that DHCP is most likely the problem. Like the solution in [this issue](https://github.com/homeworkc/lazycast/issues/3), you should disable udhcpd (i.e., delete this line https://github.com/homeworkc/lazycast/blob/fc9fcd483868071c9abe37899ac0254b573489be/all.sh#L62) and only use dnsmasq. Note that in the current release, the IP address should be ``192.168.173.80`` (instead of ``192.168.101.80``). Also, the name of the p2p interface may not always be ``p2p-wlan0-0``.
You can also use wireshark (or ``tcpdump``, see [this page](https://www.wireshark.org/docs/wsug_html_chunked/AppToolstcpdump.html)) to analyze the messages on the ``p2p-wlan0-x`` interface.

TheDeepSpacer, 2019-05-27:
Non of this works for me. This are my settings.
```
sudo nano /etc/dhcpcd.conf
	interface wlan0
	static ip_address=192.198.173.1/24
sudo service dhcpcd restart

sudo mv /etc/dnsmasq.conf /etc/dnsmasq.conf.orig
sudo nano /etc/dnsmasq.conf
	interface=wlan0
	dhcp-range=192.168.173.2,192.168.173.20,255.255.255.0,24h
	
	interface=p2p-wlan0-2
	dhcp-range=192.168.173.80,192.168.173.80,12h
	
sudo nano /etc/hostapd/hostapd.conf
	interface=wlan0
	driver=nl80211
	ssid=rasppicast
	hw_mode=g
	channel=7
	wmm_enabled=0
	macaddr_acl=0
	auth_algs=1
	ignore_broadcast_ssid=0
	wpa=2
	wpa_passphrase=adm4kit12
	wpa_key_mgmt=WPA-PSK
	wpa_pairwise=TKIP
	rsn_pairwise=CCMP
```
The acces point works fine. But when i start all.sh

Available interfaces:
p2p-dev-walan0
wlan0
Selected Interface "p2p-dev-wlan0"
FAIL

closed 2019-11-28 by TheDeepSpacer