homeworkc/lazycast #32: how do open security?

GRey-515, 2019-06-08:
what mean?
>wpa_cli set device_type 7-0050F204-1
>wpa_cli set p2p_go_ht40 1
>wpa_cli wfd_subelem_set 0 00060151022a012c
>wpa_cli wfd_subelem_set 1 0006000000000000
>wpa_cli wfd_subelem_set 6 000700000000000000

 where i can read about wfd_subelem_set arguments? is there any information anywhere about "wfd_subelem_set"?

how do "key_mgmt=NONE"?
how do open security without password and any pin at all?

homeworkc, 2019-06-08:
These are configurations defined in the [standards](https://www.wi-fi.org/discover-wi-fi/specifications).
I don't think an open wifi connection is a good idea. It may be doable but the source may not support it. (WPS PIN is mandatory in the standard so every source has to support it.) Additionally, it complicates the configuration of the DHCP server and an attacker can easily connect to the network and see the screen content, which is a big security concern.

GRey-515, 2019-06-09:
i read then "Wi-Fi Display Technical Specification Version 2.1" and don't uderstand, how "wpa_cli wfd_subelem_set 0 00060151022a012c". In this docs i find "Table 27. WFD Subelement ID Definitions", "5.1.2 WFD Device Information Subelement", read it. in this doc only 16 bits, but "00060151022a012c" have 256 bits. can you help me to understand it? thank you.

homeworkc, 2020-01-23:
It is in hex.

closed 2020-04-25 by homeworkc