homeworkc/lazycast #41: d2.py always open VLC when installed on raspbian-nspawn-64

win98se, 2019-10-26:
On [raspbian-nspawn-64](https://github.com/sakaki-/raspbian-nspawn-64), the architecture is `aarch64` instead of starting with `armv`.

[image]

This causes `d2.py` always open VLC instead of other binaries. (Line 194 of `d2.py`)

More information on architecture info on raspbian-nspawn-64 -
[image]

win98se, 2019-10-26:
In addition, using VLC, there are huge delays.

I changed line 194 of `d2.py` from -
`if (os.uname()[-1][:4] != "armv"):`
to -
`if (os.uname()[-1][:4] != "armv") and (os.uname()[-1] != "aarch64"):`

And it works.

homeworkc, 2020-02-06:
This should be fixed in the current commit.

closed 2020-02-06 by homeworkc