homeworkc/lazycast #46: huge freezes(stream from mobile phone)

arduinka55055, 2020-01-28:
i have freezes when streaming from my mobile phone(Redmi Note 8)
maybe there are some settings in config files that increase performance or something else.
thanks

homeworkc, 2020-02-03:
You could probably try different frequencies (although I found that it usually helps very little). There is also something that you may want to try. I found that sometimes it runs smoother if you replace the network system with NetworkManager (see [here](https://raspberrypi.stackexchange.com/questions/29783/how-to-setup-network-manager-on-raspbian)). However, NetworkManager messes with the DNS settings so if you are not familiar with those, backup your data (or try on a new sd card) first before you proceed.

arduinka55055, 2020-02-04:
After testing i noticed that huge freezes only in mobile games. YouTube is ok! But there are sometimes microfreezes in YouTube.

closed 2020-02-04 by arduinka55055