homeworkc/lazycast #49: Black border with every resolution

orlo11, 2020-02-05:
Hello,
I tried lazycast and everything works very well -- thanks for your work homeworkc.
Only one problem: when I cast a screen, I always get a black Border of about 10% around the picture, no matter which resolution of the screen I choose -- even with the native resolution of the screen attached to the raspi. This happens to all players (I tried no. 1/2 and the omxplaxer, always the same).
This results in poor readability of text because some pixels are droped.
Is this a bug or a feature?
I thought that omxplayer is using fullscreen,  but this seems not to be.
Best regards,
orlo11.

homeworkc, 2020-02-06:
That should not happen. What source are you using and what is the targeted resolution?

orlo11, 2020-02-06:
Thanks for the answer.
I used a fresh raspbian buster (2019-09-21), apt-updated and followed README.MD last weekend.
I used the binary zip to install.
First everything did, as I expected, but then the black border or bar showed up, whenever I used Win10 (Acer Spin 3) as source. With Android (Galaxy S7) they appeared later, when I tried with a beamer (1024x768) attatched to the Pi.
Yesterday I installed lazycast on the same image again (as now described, the git-source) and the black frame stayed. I had an older screen (1280x1024) at the Pi and the Image always used only a part in the middle -- also with the phone in portrait -- last weekend this stretched to the full screen height. 
I will try a new clean setup of raspbian at next, maybe somewhere flipped a bit wrong during I tested.

orlo11, 2020-02-07:
Today I tried a fresh install, and had the same effect. When I use a 4:3 Monitor or beamer, I have always borders or bars around the player window (even with the phone in portrait mode appears the Pi desktop above and below) with every player.
When I use a 16:9 screen at the Pi then the picture from my phone fills the screen completely, when in landscape and stretched from top to bottom in portrait mode.
But from Win 10 (19.09) there always appears a black border. 
When I use player 1 or 3 then then I saw the following:
I adjusted the screen resolution on the notebook to the native one of the screen attached to the Pi before I connected. Then the notebook switched to 1920 x1080 right in that moment when the picture showed up on the Pi screen. When I closed the connetion it returned to the lower resolution.
When I change the resolution after I connected, then the Pi chokes sometimes (white  unnamed window) but worked again, when I switched the resolution some times back and forth.
Do the Pi and the PC talk about the resolutions, they can handle?
Maybe there are misunderstandings, that lead to the fact, that the Windows machine thinks it should send a Full-HD stream, instead of one the lower resolution.

homeworkc, 2020-02-08:
The codes related to resolutions are this line:
https://github.com/homeworkc/lazycast/blob/24c4c43ad16f17e92d79fa41688388a77e2860cd/d2.py#L92
and 
this line:
https://github.com/homeworkc/lazycast/blob/24c4c43ad16f17e92d79fa41688388a77e2860cd/d2.py#L94

Specifically, each bit in this string ``0001FFFF 3FFFFFFF 00000FFF`` corresponds to a certain resolution with a specific refresh rate.
Currently there is ``lazycast`` does not detect the native resolution of the screen connected to Pi and simply tells the source it supports any resolution. That's probably why a Windows machine chooses full HD. 
You force a source to only use a certain resolution by specifying it in this string. The mapping table can be found in the standards. For 1024x768, try ``00000000 0000000C 00000000``. For 1280x1024, try  ``00000000 0000C000 00000000``

orlo11, 2020-02-09:
Thank you very much, I'll try it tomorrow.

orlo11, 2020-02-10:
Hello, your advice showed me the way to find a solution that works with Android. Thank you.
With my phone there are no black borders more, if I provide the correct resolutions, the display is able to show but from Windows the remained (or even worse).
I downloaded the paper from the WIFI-alliance and used the tables to offer only the correct resolutions from d2.py. When I used the omxplayer then I could see in the terminal, that windows asked for the correct resolution, but the displayed picture has always a border of about 100 pixels around. The Android-phone asked for the same resolution, but with @30Hz instead of @60Hz from Win.
Then I had the possibility to test from another Windows 10 notebook (faster with an screen res. of 3200xsomewhat). This failed nearly complete (although I reduced the resolution to a displayable size). Sometimes I got a picture first, until in Win the message popped up to enable the mouse/keyboard backchannel from the Pi (although I had this switched off in d2.py), then the connection broke.
Most times the connection broke, when the picture on the Pi was expected to show up and often then d2.py was hooked up. I had to kill the terminal-window.
It does no matter, which player I used.
During the rest of the week I will have no time to test further, so I'll try on next weekend.

orlo11, 2020-02-20:
Hello again,
I close this issue because it showed, that it is a Windows problem, not one of lazycast.
Yesterday I had the possibillity to test the cast from my Windows to a Samsung UE.... tv, and
there I discovered the same black border.
Again I know why I don't like Windows ... 
Thanks for your afford.

closed 2020-02-20 by orlo11

piotr-cz, 2025-02-04:
In my client only these resolutions were available:
- 1280 x 720
- 1360 x 768
- 1366 x 768
- 1600 x 900
- 1920 x 1080

This is how I enabled support for 1680 x 1050 resolution that was not listed in client by editing [dy.py](https://github.com/homeworkc/lazycast/blob/master/d2.py#L220-L223) file:

```diff
- if disable_1920_1080_60fps == 1:
- 	msg = msg + 'wfd_video_formats: 00 00 02 10 0001FEFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none\r\n'
- else:
- 	msg = msg + 'wfd_video_formats: 00 00 02 10 0001FFFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none\r\n'
+ msg = msg + 'wfd_video_formats: 00 00 02 10 00000000 3FFFFFFF 00000FFF 00 0000 0000 00 none none\r\n'
```

Changing last line to this one also works:

```diff
+ msg = msg + 'wfd_video_formats: 00 00 02 10 00000000 0C000000 00000000 00 0000 0000 00 none none\r\n'
```

After changes, available resolutions are:
- 1280 x 768
- 1280 x 800
- 1440 x 900
- 1680 x 1050

Bits are described in
- *Wi-Fi Display Technical Specification Version 2.1* document
- and here: https://github.com/benzea/miracast/blob/master/rtsp.py