homeworkc/lazycast #50: difficulties using NetworkManager

Brickabrak, 2020-02-07:
First off, thank you for your efforts to make this a useful casting app. I have high hopes that I can get it working. I've tried this install a couple of different ways. Using an RPi CM3+ with 2 wireless dongles (edup ep-n8508 and Edimax EW-7611ULB), an RPi3A+ with built-in wifi.
I typically have been using SSH to connect and install things. 
 
I've tried this first using Raspbian Buster Lite with no success. Essentially wifi would not longer work after the 
>"sudo apt install network-manager network-manager-gnome openvpn openvpn-systemd-resolved network-manager-openvpn network-manager-openvpn-gnome" 
>and 
>"sudo apt purge openresolv dhcpcd5" commands. 

I was able to run the all.sh with ethernet, but got the same error posted here https://github.com/homeworkc/lazycast/issues/43#issue-523750242

I tried this on an existing system and with a fresh image, on the CM3+ and RPi3A+, wifi would fail around 89% finished with installing and wlan0 could no longer be found. 

Using a fresh image of "Raspbian Buster Desktop with recommended software" I tried skipping those steps and could get to all.sh giving me a pin code, but I couldn't find the device on the network, so I went and installed network manager which again, kills wlan0 around 89% finished with the install. 
Going into Advanced Network Settings in the Raspbian desktop I can then make a connection called the same name as the all.sh script will name it, make that a hotspot using the wlan0 device, and Cloned MAC address = "Stable" 
I can also use a USB to Ethernet adapter to keep an internet connection on the device. After rebooting it looks like I can connect to the device as a hotspot, but the password is the one I have it in Advanced Network Settings, not a pin. 
Also after that reboot, all.sh fails:
pi@Streamer:~/lazycast $ ./all.sh
Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK

Selected interface 'p2p-dev-wlan0'
FAIL
Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
FAIL
Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
FAIL

The only way I can get a pin is by removing the Hotspot connection and letting wlan0 either just scan or be connected to an SSID. 
There must be something I'm missing here. is there anything obvious from what I've mentioned that could be causing this problem?

homeworkc, 2020-02-08:
I suggest test on RPi3A+ with Raspbian Buster Lite and a connected monitor first. When installing NetworkManager, it resets the network and stored credentials. Therefore, ``ssh`` would not work after that because you need to configure the network on Pi locally for it to connect to the network again.
If you must use ``ssh``, then I'll suggest follow the steps and skip the NetworkManager part. You may need to wait a while for Pi to set up a p2p interface when you run ``./all.sh`` but it should work nevertheless.

amerinoj, 2020-02-11:
The wlan0 integrated adapter in RP3B+ only can work in one mode , you can't use the same adapter to P2P and client or hotspot

when p2p-dev-wlan0 is created and managemened by WPA_SUPPLICANT the father interface change the mode because the NETWORK_MANAGER try to connect to Wireles infrastruture.

check the main interface state when P2p interface is created:
iw dev wlan0  info