homeworkc/lazycast #52: Support for Raspberry Pi 3A?

smoonlee, 2020-02-22:
Have downloaded the repo onto my Rpi 3A but when I try can connect either my laptop or tablet it just say failed to connected.

homeworkc, 2020-02-23:
Please try downgrading the ``wpa_supplicant`` package.

homeworkc, 2020-02-23:
Also, a reboot is needed after downgrading the package.

smoonlee, 2020-02-23:
Hi @homeworkc thanks for the reply, rebuilding my Pi3A and will test it shortly just waiting for some updates to complete.

smoonlee, 2020-02-23:
so it connects then a couple of seconds later disconnected me

smoonlee, 2020-02-23:
Ha just managed to get it to connect and show my display then it disconnces again

smoonlee, 2020-02-23:
tried same image ona 3B+ and doesnt the same with latpop and tablet connects and instantly disconnects

homeworkc, 2020-02-23:
What is the message after disconnection? Is it ``recv timeout`` or other teardown messages?

smoonlee, 2020-02-23:
[image]

Not the best image but that's what I get after disconnect

homeworkc, 2020-02-23:
Can you try using ``player_select = 2``? I think this error has something to do with the encoding of audio streams. (Player 1 accepts AAC while player 2 just uses plain PCM.)

smoonlee, 2020-02-28:
Sure, I've got some time tomorrow will give it another go, apologies for the delay in replies

smoonlee, 2020-04-26:
@homeworkc  Do you recommend Raspbian Lite or Desktop for this? Got a new SD Card, unrelated to this initial issue, so thought id do some more testing as a 3A would make an ideal little unit for this

homeworkc, 2020-04-30:
The image I always used and tested is "Raspbian Buster with desktop" from [here](https://www.raspberrypi.org/downloads/raspbian/). (The 1136MB one.)

sibav, 2020-06-14:
Did anyone figure this out? Tried to print out the error. Seems Errno 11 Resource temporarily unavailable.

closed 2022-11-30 by smoonlee