homeworkc/lazycast #54: Win 10: Connection couldn't be established

Strajuck, 2020-03-14:
My Windows 10 source, recognizes the raspberry as a wireless display, but it can't establish a connection between them. Can anyone tell me what the issue is?

I followed all the advice given in the readme.

Regards and thanks for your help in advance

Strajuck

165plo, 2020-03-19:
@Strajuck Ensure that BusyBox is installed on your pi. This was an issue for me.
`sudo apt-get install busybox`

homeworkc, 2020-03-28:
What is the output of the terminal? What is your system (Pi 3 a/b/b+ or 4, using built-in wifi)? Did you downgrade the wpasupplicant package?

Strajuck, 2020-03-29:
@165plo The most recent version of Busybox is installed on my pi.

@homeworkc The terminal displays: Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-dev-wlan0'
OK
5	DIRECT-uF	aa:4d:54:d4:c6:85	[DISABLED][P2P-PERSISTENT]
=5	
Selected interface 'p2p-dev-wlan0'
OK
Selected interface 'p2p-wlan0-0'
Available interfaces:
p2p-wlan0-0
p2p-dev-wlan0
wlan0
Selected interface 'p2p-wlan0-0'
Available interfaces:
p2p-wlan0-0
p2p-dev-wlan0
wlan0
p2p-wlan0-0
The display is ready
Your device is called: lazycast_on_raspberrypi
PIN:
31415926

I am also capable of selecting the device on my windows 10 computer. But if I enter the passcode, nothing happens. I am using a Raspberry Pi 3b with integrated wifi. And I did reboot and downgrade wpasupplicant.

homeworkc, 2020-03-29:
Okay, here is another test. Run ``all.sh``. After the PIN number is shown, open another terminal and run ``sudo wpa_cli``. Then try to select the device on the computer and see what the output of wpa_cli is.

Strajuck, 2020-03-29:
It just says "Selected interface 'p2p-wlan0-2' and "Interactive mode". There is no other output given...

homeworkc, 2020-03-29:
What about running ``sudo wpa_cli`` first then run ``all.sh``? 
Also, I found that sometimes installing ``NetworkManager`` helps. You can find the instruction [here](https://github.com/homeworkc/lazycast/tree/c5762ec1b34a2d8851d6a723c422b891779ea1a6). If you find that your DNS does not work (it may take a while for this to happen) after installing NetworkManager, try ``sudo systemctl stop systemd-resolved``. (See [here](https://unix.stackexchange.com/questions/518266/ping-displays-name-or-service-not-known) for details.)

Strajuck, 2020-03-30:
If i proceed with sudo wpa_cli first and then run all.sh it outputs the following: 

Interactive mode

<3>CTRL-EVENT-SCAN-STARTED 
<3>P2P-FIND-STOPPED 
<3>P2P-GROUP-STARTED p2p-wlan0-0 GO ssid="DIRECT-uF" freq=2462 passphrase="vpuVvXMP" go_dev_addr=22:e4:94:12:df:83 [PERSISTENT]
<3>CTRL-EVENT-SCAN-RESULTS 
<3>CTRL-EVENT-SCAN-RESULTS 
> 
Installed the other network manager. 

And it worked. Thanks a lot for helping me fix the Issue, i really appreciate your help!

closed 2020-03-30 by Strajuck