homeworkc/lazycast #56: Cannot connect to Rpi3 display with Windows 10

Dittizardichu, 2020-03-26:
When I attempt to connect to my lazycast Rpi3, I get the error "couldn't connect to your device." I was hoping to change the interface the lazycast uses for its network to see if that would help, but I cannot find a way to change the interface.

thomaswjholland, 2020-03-27:
Hey - I have the same issue - have you found a fix?

homeworkc, 2020-03-28:
Couple of questions:
Are you using the built-in wifi card?
Is there any other network connection going on?
What is the system image you use?
Did you downgrade the wpasupplicant package?
Did you enter the PIN (31415926)? If you're pairing it for the first time you need to enter the PIN. Windows 10 sometimes does not prompt you to enter the PIN but if you click on the ``CONNECT`` tab again you will see that it is still waiting for the PIN.

homeworkc, 2020-03-28:
Also, you need to reboot after downgrading wpasupplicant for the correct version to run.

pkbaharani, 2020-04-16:
same problem, unable to connect from windows 10