homeworkc/lazycast #61: Android cast hangs on "connecting" state.

Istria1704, 2020-04-09:
Hello.

I found your lazycast tool and it suits my need perfectly. I want to connect my Pi3 to my ceiling mounted projector and use it as a wireless display for my Windows PC. 

I didn't get it to work using Buster, having the same problem already mentioned not having Internet access anymore after downgrading. So I just used an older (stretch) Raspbian rom. Now it works up until the password is printed to the screen every couple of seconds.

In Windows 10, however, it says my PC does not support wireless displays (the laptop is 10 years old). 
On my phone (Android), the lazycast does show up in the list of wireless displays, but when I click it it says "connecting" for a while (20s or so) and then stops. I am also never asked for the password.

Is it supposed to work on Android? Of is it Windows only? 

After fiddling for 2 hours I took a break and went out to get some takeout. And guess what, in the restaurant  there is a TV in the corner saying "connect your screen wirelessly". Tried it, and it worked on the first try. It says "connecting" for 3-5s, and then it's connected. 

Any idea?

homeworkc, 2020-04-09:
For Windows 10, the processor on a PC needs to be newer than 3rd gen core processors. (i.e., at least as new as Ivy Bridge.)
It does work with Android. It works with a Samsung S6 Edge without any issue. What is the output of the terminal? What phone are you using and what is the version of Android?
Also, an alternative to downgrading is to compile the newest version of wpa_supplicant yourself. You can find the instruction in the [MICE](https://github.com/homeworkc/lazycast#miracast-over-infrastructure-under-development) section. (Just run all the commands and reboot.)

Istria1704, 2020-04-09:
Thanks for the swift response.

The Windows 10 PC is a Core2Duo SP9400 piece of antique I bought during my first year of university 10 years ago. I will have to upgrade sooner rather than later anyway. So that will not be a problem.

When I try to connect from my Xiami Mi Mix 2S (rooted and running a LineageOS rom) there is no difference on the Pi screen. It just keeps spitting out the password every couple of seconds. No change when I try to connect.

homeworkc, 2020-04-09:
One way to check the connection is to open another terminal and run" ``sudo wpa_cli``. Then try to connect using the smartphone.
Is it possible that the restaurant TV is actually running Chromecast and not Miracast? Android 9 might have very bad support for Miracast if the vendor does not support it themselves.

Istria1704, 2020-04-09:
I'll try that in a sec. 

If it doesn't work, what Android version is your S6 on? I might flash that to my Nexus 5X to see if that makes a difference.

closed 2020-04-09 by Istria1704

Istria1704, 2020-04-09:
Tried the wpa_cli thing in a second terminal. It responds to my phone trying to connect. See the photo below. 

On the phone side, it's the same as earlier. Says "connecting" for a while. And then stops. 

[image]

homeworkc, 2020-04-09:
My S6 Edge runs Android 7.

homeworkc, 2020-04-09:
It says "AP-STA-CONNECTED", which means wpa_supplicant is not an issue.

homeworkc, 2020-04-09:
So the problem might be: udhcpd or the python script. Do you have any external network interface on you Pi?

Istria1704, 2020-04-09:
Nope. It's a stock Pi 3 using the onboard WiFi. Nothing connected but power, hdmi and a USB dongle for mouse and keyboard.

By the way. I noticed something with the wpa_cli:
When I click on the discovered lazycast "wireless display" on my Android, the Pi output says "AP-STA-CONNECTED after a couple of seconds. It stays like that for about 20s. The exact same moment that on my phone it stops saying "connecting", on the Pi, it gives a new line saying AT-STA-DISCONNECTED. Not sure if that is to be expected or not.

homeworkc, 2020-04-09:
Yeah. It is the expected behavior. Basically after "AP-STA-CONNECTED", Pi should assign IPs to itself (``192.168.173.1``) and to the device (``192.168.173.80``)  using DHCP. Then Pi will try to connect to  ``192.168.173.80`` port 7236. It seems like the main problem is this connection process. What happen if you try to ping ``192.168.173.80`` after "AP-STA-CONNECTED"?

homeworkc, 2020-04-09:
You could also try using ``adb`` and run ``ifconfig`` on the smartphone side after "AP-STA-CONNECTED" to see whether it gets ``192.168.173.80`` or not.

Istria1704, 2020-04-09:
I pinged the ip address from a third terminal after CONNECTED and before DISCONNECTED. Result below.

[image]

Istria1704, 2020-04-09:
Also tried ifconfig via adb while "connecting". It seems to still have the ip received by my routers dhcp (192.168.1.16).

[image]

homeworkc, 2020-04-09:
I couldn't remember planting any easter egg haha. Did I forgot to delete some temporary stuff? Also, I think the problem here is the DHCP server, which is associated with this line: https://github.com/homeworkc/lazycast/blob/839649e7005accc239b0820f6ce56246b806117b/all.sh#L65

homeworkc, 2020-04-09:
The IP of wlan0 should be the same. Do you see any interface like ``p2p-wlan0-x``? (Might be the entry before p2p0)

homeworkc, 2020-04-09:
You could also try installing udhcpd package manually (by apt install) and replace the line with ``sudo udhcpd ./udhcpd.conf``, although I am surprised that the dhcp server didn't work since I didn't run into any dhcp issue with all the devices I have.

Istria1704, 2020-04-09:
Nope. Nothing like that. p2p0 is the only p2p in the whole thing.

About the sandbox thing. I do see this:
[image]

sudo sandbox: command not found.

homeworkc, 2020-04-09:
Um it's busybox. Try run ``busybox`` in the terminal and see whether ``udhcpd`` in listed in the "Currently defined functions".

homeworkc, 2020-04-09:
Oh I didn't know that before.

Istria1704, 2020-04-09:
> Um it's busybox. Try run `busybox` in the terminal and see whether `udhcpd` in listed in the "Currently defined functions".

Yes, I fixed it in the post above. The brain is tired I guess. =p

Istria1704, 2020-04-09:
Running "busybox" in the terminal gives:

-bash: busybox: command not found

homeworkc, 2020-04-09:
What? That's interesting. Try installing then. (``sudo apt install busybox``)

Istria1704, 2020-04-09:
Works now after installing busybox and rebooting. Hooray! 

[image]

homeworkc, 2020-04-09:
Cool. What is the OS image running on your Pi? I thought busybox is installed on most distributions.

Istria1704, 2020-04-09:
I used raspbian stretch lite from 08-apr-2019. It was described as the "latest stretch version".

It doesn't work perfectly yet by the way. Here is a video link:
https://drive.google.com/file/d/11Uba2ZnLVGRUeH30FTtWe2eewHJd4-fu/view?usp=drivesdk

The wireless router is 2,5m away without walls.

Istria1704, 2020-04-09:
I see I screwed up the video permissions. Should be viewable now by "anyone with a link".

homeworkc, 2020-04-09:
Interesting. How about using omxplayer or player1? (Modify this line: https://github.com/homeworkc/lazycast/blob/839649e7005accc239b0820f6ce56246b806117b/d2.py#L27)

Istria1704, 2020-04-09:
The player/select is already set to 2 by default.

homeworkc, 2020-04-09:
Yeah. You can change it. (``player_select = 3``)

Istria1704, 2020-04-09:
I'm getting some errors in mode 3 when connecting:

https://drive.google.com/file/d/11Uba2ZnLVGRUeH30FTtWe2eewHJd4-fu/view?usp=drivesdk

Istria1704, 2020-04-09:
[image]

homeworkc, 2020-04-09:
How about ``player_select = 1``? This could either be video codec or network issues. You could also try different smartphone and see whether it gives different results.

Istria1704, 2020-04-09:
Apparently, omxplayer was not installed also. After installing it, mode 3 works. But with the same flickering/stuttering image. The audio is perfect though. 

I'll try mode 1 now.

homeworkc, 2020-04-09:
The Raspberry Pi image I used is "Raspbian Buster with desktop
Image with desktop based on Debian Buster" from [here](https://www.raspberrypi.org/downloads/raspbian/), so if you have a spare SD card I would suggest try this image with instructions in README.

Istria1704, 2020-04-09:
Same thing in mode 1. =(

Only difference is the latency. In mode 1 it's around 200ms. In mode 3, it was around 3 seconds.

homeworkc, 2020-04-09:
It also might be the case where the encoding of video on the smartphone side does not conform to the standards so it's worth to try other devices available to you.

Istria1704, 2020-04-09:
I will start again from zero tomorrow using Buster with desktop image. Should I just use the latest version from the official raspbian website?

Have a good night/day. It's late over here, so time to sleep for me. Thanks a lot for the superfast help so far!

homeworkc, 2020-04-09:
Yes. Specifically, [this](https://downloads.raspberrypi.org/raspbian_latest) direct link from the website.

Istria1704, 2020-04-09:
> Yes. Specifically, [this](https://downloads.raspberrypi.org/raspbian_latest) direct link from the website.

I see you linked it before also. See, I'm tired. Goodnight. =)

Istria1704, 2020-04-10:
Hey,

I started fresh with the raspbian image from your link.

1. Followed the instructions in the MICE section.
2. Installed lazycast.

It works, but still with the same stuttering/flickering of the video. Auto is perfect.
https://drive.google.com/file/d/11k7ix8apgwhU-cEWiA9WCR9lmeLZqBEg/view?usp=sharing

I was thinking, could it be that my phone does not have a 1920x1080 resolution, but 2160x1080? I read that 1920x1080@60Hz is the maximum supported resolution.

Istria1704, 2020-04-10:
Also, I now downgraden my Nexus 5X to a stock 7.0 rom. But the lazycast does not show up in the list.

On the Xiaomi with Android 9, there is a checkbox I have to tick "wireless displays" before it shows up. On Android 7 that option is not there. It just doesn't see the lazycast.

homeworkc, 2020-04-10:
The "wireless displays" checkbox can be enabled by setting ``persist.debug.wfd.enable=1`` in the ``build.prop`` file (if rooted). I would suggest try Nexus 5X first.
What is the output of the terminal? The resolution (and also the profile of the h264 encoding) might contribute to the issue. In my case, the S6 automatically scales the wireless screen to 1080p. These information would be included in the ``wfd_video_formats`` string.

Istria1704, 2020-04-10:
Here is the console output until disconnecting from the Android side:

`pi@raspberrypi:~/lazycast $ ./all.sh
Selected interface 'p2p-wlan0-0'
Available interfaces:
p2p-wlan0-0
p2p-dev-wlan0
wlan0
already on
p2p-wlan0-0
The display is ready
Your device is called: lazycast_on_raspberrypi
PIN:
31415926
---M1--->
OPTIONS * RTSP/1.0
CSeq: 0
Require: org.wfa.wfd1.0

<--------
RTSP/1.0 200 OK
CSeq: 1
\Public: org.wfa.wfd1.0, SET_PARAMETER, GET_PARAMETER

<---M2---
OPTIONS * RTSP/1.0
CSeq: 100
Require: org.wfa.wfd1.0

-------->
RTSP/1.0 200 OK
CSeq: 100
Public: org.wfa.wfd1.0, GET_PARAMETER, SET_PARAMETER, SETUP, PLAY, PAUSE, TEARDOWN

---M3--->
GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 1
Content-Length: 130
Content-Type: text/parameters

wfd_audio_codecs
wfd_video_formats
wfd_client_rtp_ports
wfd_uibc_capability
wfd_standby_resume_capability
wd_initial_buffer

<--------
RTSP/1.0 200 OK
CSeq: 2
Content-Type: text/parameters
Content-Length: 484

wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play
wfd_audio_codecs: AAC 00000001 00
wfd_video_formats: 00 00 02 10 0001FEFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none
wfd_3d_video_formats: none
wfd_coupled_sink: none
wfd_display_edid: none
wfd_connector_type: 05
wfd_uibc_capability: input_category_list=GENERIC, HIDC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=Keyboard/USB, Mouse/USB;port=none
wfd_standby_resume_capability: none
wfd_content_protection: none

---M4--->
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 2
Content-Length: 406
Content-Type: text/parameters

wfd_audio_codecs: AAC 00000001 00
wfd_video_formats: 00 00 02 10 00000080 00000000 00000000 00 0000 0000 00 none none
wfd_presentation_URL: rtsp://255.255.255.255/wfd1.0/streamid=0 none
wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play
wfd_uibc_capability: input_category_list=GENERIC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=Keyboard/USB, Mouse/USB;port=4321
wfd_uibc_setting: enable

<--------
RTSP/1.0 200 OK
CSeq: 3

uibcport:4321

---M5--->
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 3
Content-Length: 27
Content-Type: text/parameters

wfd_trigger_method: SETUP

<--------
RTSP/1.0 200 OK
CSeq: 4

<---M6---
SETUP rtsp://192.168.173.80/wfd1.0/streamid=0 RTSP/1.0
CSeq: 101
Transport: RTP/AVP/UDP;unicast;client_port=1028

-------->
RTSP/1.0 200 OK
CSeq: 101
Date: Fri, Apr 10 2020 14:51:57 GMT
Session: 00000012
Transport: RTP/AVP/UDP;unicast;client_port=1028;server_port=19022

['RTSP/1.0 200 OK\r\nCSeq: 101\r\nDate: Fri, Apr 10 2020 14:51:57 GMT\r\nSession: 00000012\r\nTransport: RTP/AVP/UDP', 'unicast', 'client_port=1028', 'server_port=19022\r\n\r\n']
['server_port=19022\r\n\r\n']
19022
<---M7---
PLAY rtsp://192.168.173.80/wfd1.0/streamid=0 RTSP/1.0
CSeq: 102
Session: 00000012

-------->
RTSP/1.0 200 OK
CSeq: 102
Date: Fri, Apr 10 2020 14:51:57 GMT

---- Negotiation successful ----
have a nice day ;)
Video codec omx-h264 width 1920 height 1080 profile 100 fps 30.000000
Audio codec aac channels 2 samplerate 48000 bitspersample 16
Subtitle count: 0, state: off, index: 1, delay: 0
V:PortSettingsChanged: 1920x1080@30.00 interlace:0 deinterlace:0 anaglyph:0 par:1.00 display:0 layer:0 alpha:255 aspectMode:0
GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 4
Session: 00000012

['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 4\r\nSession: 00000012', '']
['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 4\r\nSession: 00000012']
RTSP/1.0 200 OK
CSeq: 4

GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 5
Session: 00000012

['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 5\r\nSession: 00000012', '']
['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 5\r\nSession: 00000012']
RTSP/1.0 200 OK
CSeq: 5

GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 6
Session: 00000012

['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 6\r\nSession: 00000012', '']
['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 6\r\nSession: 00000012']
RTSP/1.0 200 OK
CSeq: 6

GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 7
Session: 00000012

['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 7\r\nSession: 00000012', '']
['GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0\r\nCSeq: 7\r\nSession: 00000012']
RTSP/1.0 200 OK
CSeq: 7

SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 8
Session: 00000012
Content-Length: 30
Content-Type: text/parameters

wfd_trigger_method: TEARDOWN

Terminated
`

homeworkc, 2020-04-10:
It seems that the smartphone is sending the highest h264 profile at 1080p. How about changing ``02 10`` in this line 
https://github.com/homeworkc/lazycast/blob/839649e7005accc239b0820f6ce56246b806117b/d2.py#L95
to ``01 01``?

Istria1704, 2020-04-10:
No joy. Same result. Here is the new log.
[console after 01 01 change.txt](https://github.com/homeworkc/lazycast/files/4462166/console.after.01.01.change.txt)

homeworkc, 2020-04-10:
Yeah. Sound like a good idea. It's really weird since I can get the a Windows 10 PC working with the exact parameter (wfd_video_formats: 00 00 02 10 00000080 00000000 00000000 00 0000 0000 00 none none) without an issue.

Istria1704, 2020-04-10:
I managed to enable the miracast function on the Nexus 5x. I just can't get some luck:

The lazycast shows up on the phone. When I click it, the pi console reaches "NEGOTIATION SUCCESFUL". But them starts printing out the password again. But this time faster (1x per second).

On the phone, it says "connected".

homeworkc, 2020-04-10:
Have you tried different player_select?

Istria1704, 2020-04-10:
Tried 1 and 3 with the Nexus 5X.

homeworkc, 2020-04-10:
How about 2?

Istria1704, 2020-04-10:
Same in 1, 2 and 3. After negotiation succesful, it starts printing out the password again. But no video screen pops up.
Phone stays "connected".

homeworkc, 2020-04-10:
What is the message in the terminal? Maybe 5X tries to use a different port than 1028.

Istria1704, 2020-04-10:
Here's the log.
[nexus5x log (player mode 3).txt](https://github.com/homeworkc/lazycast/files/4462635/nexus5x.log.player.mode.3.txt)

Istria1704, 2020-04-10:
huh, I just noticed I'm Android 8.1... Not 7.
Not sure how that happened.

homeworkc, 2020-04-10:
Also, what is the output if you run ``tvservice -s``? The resolution string in ``d2.py`` could match the projector you're using and maybe it would help.

homeworkc, 2020-04-10:
The trace of 5X seems to be normal except ``d2.py`` exits without a reason. What if when 5X says connected, open another terminal and run ``omxplayer rtp://0.0.0.0:1028 --live``?

homeworkc, 2020-04-10:
Also if you want to flash the rom again, try to find the earliest firmware available. I think Google tried to phase out Miracast gradually so every major OS update they took away the relevant libraries one by one.

Istria1704, 2020-04-10:
[image]

homeworkc, 2020-04-10:
Okay. Try using ``wfd_video_formats: 00 00 01 01 00000000 00000008 00000000 00 0000 0000 00 none none`` with Android 9 and see what happens. It also worth trying  ``wfd_video_formats: 00 00 01 01 00000001 00000000 00000000 00 0000 0000 00 none none``.

Istria1704, 2020-04-10:
On the Xiaomi you mean? Not Nexus?

[5 more replies]