homeworkc/lazycast #63: Can't connect after NetworkManager

LiveITA, 2020-04-11:
Hi, i'm setting my raspi in headless mode, so once i follow the NetworkManager step, my raspi is gone for good. What can I do to keep working in ssh?
Thanks

homeworkc, 2020-04-11:
Do you have a monitor that you can connect to? (Note that lazycast does not forward the screen content via ssh so you will need a monitor.) After you connect the monitor, you can re-connect to the network (either wireless or wired) using the NetworkManager icon on the upper right corner. ssh should work after re-connection.

LiveITA, 2020-04-11:
ThanksThat worked! It's going smoothly!Just 2 questions:Can I remove the pin entirely?Can I use a 5ghz USB dongle and disable the integrates wifi?

homeworkc, 2020-04-11:
1. Other people and I have tried using alternative methods other than PIN but we found that PIN is the most reliable way of setting up a connection. (Also it's the only mandatory method in the specification, which means that alternative methods may not be supported by PC or smartphones.)
2. It depends on the driver support of the USB dongle. The easiest way to check is to run: ``sudo wpa_cli interface`` and see whether the external wifi has a ``p2p-dev-*`` interface. Note that if your Pi is newer than 3B+, 5Ghz wifi is supported and you can manually specify the p2p frequency in this line: https://github.com/homeworkc/lazycast/blob/9ecfdb91f0ddd5c0c9746b9fce3dd1f4a82cf344/all.sh#L54