homeworkc/lazycast #64: Latest commit seems to break on Ubuntu 18.04

leopck, 2020-04-15:
I saw your latest commit [https://github.com/homeworkc/lazycast/commit/9ecfdb91f0ddd5c0c9746b9fce3dd1f4a82cf344](https://github.com/homeworkc/lazycast/commit/9ecfdb91f0ddd5c0c9746b9fce3dd1f4a82cf344) this commit is breaking on Ubuntu 18.04. After this commit onwards all initialization of my P2P WiFi constantly fails.

I had to revert one commit behind to get it to work again. Image shown below.

[image]

Any idea what's going on?

leopck, 2020-04-15:
Okay, after debugging some lines of codes inside the all.sh this is the issue 
`sudo wpa_cli p2p_group_add -i$p2pdevinterface persistent$perstr $wlanfreq`
https://github.com/homeworkc/lazycast/commit/9ecfdb91f0ddd5c0c9746b9fce3dd1f4a82cf344#diff-538fac8ab28d29a0c13a22c3409f8b99R53

For me, my tablet is executing this
`sudo wpa_cli p2p_group_add -iwlp1s0 persistent=0 freq=5745`

This shows "Fail" as soon as I ran this. However, if I removed the interface and frequency, this error stops appearing, that's really weird. I wonder if the WPA_CLI is different.

```
sudo wpa_cli p2p_group_add -iwlp1s0 persistent=0 freq=5745 #fail
sudo wpa_cli p2p_group_add -iwlp1s0 persistent=0 #fail
sudo wpa_cli p2p_group_add persistent=0 freq=5745 #fail
sudo wpa_cli p2p_group_add persistent=0 #works
```

leopck, 2020-04-15:
My wpa_cli version
```
sudo wpa_cli -v       
wpa_cli v2.6
Copyright (c) 2004-2016, Jouni Malinen <j@w1.fi> and contributors
```

homeworkc, 2020-04-15:
Thanks for letting me know. I modified ``all.sh`` in recent commit because I want to make the p2p interface use the same frequency as the wlan interface. Maybe we can add some ``if`` block here so that if the command fails it will fall back to the last one.

leopck, 2020-04-16:
We could, though I'm also wondering myself, why does this function fail? does wpa_cli not recognize -i and freq?

homeworkc, 2020-04-21:
Okay. In the latest commit I added an option called ``managefrequency`` in ``all.sh``. Also, it will be automatically set to ``0`` if ``p2p_group_add`` with ``freq=`` fails.

leopck, 2020-04-22:
awesome, verified to work on Ubuntu 18.04. Tomorrow once, Ubuntu 20.04 is out, I'll be testing this on Ubuntu 20.04.

Thanks for all the great work. :)

closed 2020-04-22 by leopck