homeworkc/lazycast #65: Ubuntu 20.04 wpa_cli 2.9 doesn't start

leopck, 2020-04-23:
Just installed Ubuntu 20.04 and now I'm trying lazycast, but it seems that Ubuntu 20.04 is using wpa_cli v2.9 and it doesn't seem to start, it just keeps looping without giving the pin.

[image]

leopck, 2020-04-25:
After some investigation, it's really weird. 

`result=$(sudo wpa_cli p2p_group_add -i$p2pdevinterface persistent$perstr $wlanfreq)`

Result shows OK, however, `sudo wpa_cli interface`, doesn't show the p2p-wlp1s0, only the devinterface. I've left it to run multiple times, and rebooted but still doesn't seem to add the p2p-wlp1s0 only the dev interface.

```
$ sudo wpa_cli interface
Selected interface 'p2p-dev-wlp1s0'                     
Available interfaces:                                                   
p2p-dev-wlp1s0                                                            
wlp1s0
```

homeworkc, 2020-04-25:
I experimented it a little bit and it seems like the problem is not easy to solve. It seems that Ubuntu does not create the GO interface either using ``wpa_cli`` or dbus interface (which ``mice.py`` uses). It may worth trying recompiling wpasupplicant (the relevant commands are  in ``command.txt``) or seeing the debugging messages from wpasupplicant (some commands are in ``logging.sh``).

leopck, 2020-04-26:
Tried recompiling based on the command.txt, still doesn't work.... Hmm.... I've been spending a couple of days debugging this issue, but I running out of ideas.

leopck, 2020-04-26:
Sounds silly, but I've also downgraded my Ubuntu 20.04 wpa supplicant to the version that you stated in the README.md

`wpasupplicant_2.4-1+deb9u4_amd64.deb`

However, this also still shows the same behaviour as the wpasupplicant_2.9, it also doesn't work. :(

waihuling, 2024-06-27:
Hello, I also met the same problem, on 22.04, 2024 today, is there a solution to this problem