homeworkc/lazycast #76: make errors on a pi zero

idimou, 2020-09-29:
Dear homeworkc, 
good idea and it looks like an active maturing implementation. Thanks for all the effort to provide such a useful solution.

I've followed the steps to configure lazycast on a pi zero. I've connected via shh over ethernet (zero also connected to my WLAN) and when I clone and make lazycast I get a bunch of errors:

`idimou@zero:~/lazycast $ make
make -C player/.
make[1]: Entering directory '/home/idimou/lazycast/player'
cc -DOMX_SKIP64BIT -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont   -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont -g -c player.c -o player.o -Wno-deprecated-declarations
player.c: In function ‘copy_into_buffer_and_empty’:
player.c:221:6: warning: unused variable ‘buff_size’ [-Wunused-variable]
  int buff_size = buff_header->nAllocLen;
      ^~~~~~~~~
player.c: In function ‘setup_demuxer’:
player.c:470:50: warning: comparison of constant ‘0’ with boolean expression is always false [-Wbool-compare]
   if (!avcodec_open2(codec_context, codec, NULL) < 0)
                                                  ^
player.c:470:50: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
player.c:470:7: note: add parentheses around left hand side expression to silence this warning
   if (!avcodec_open2(codec_context, codec, NULL) < 0)
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       (                                         )
player.c:491:50: warning: comparison of constant ‘0’ with boolean expression is always false [-Wbool-compare]
   if (!avcodec_open2(codec_context, codec, NULL) < 0)
                                                  ^
player.c:491:50: warning: logical not is only applied to the left hand side of comparison [-Wlogical-not-parentheses]
player.c:491:7: note: add parentheses around left hand side expression to silence this warning
   if (!avcodec_open2(codec_context, codec, NULL) < 0)
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       (                                         )
player.c: In function ‘setup_clockComponent’:
player.c:602:23: warning: unused variable ‘clock’ [-Wunused-variable]
     OMX_COMPONENTTYPE*clock = ilclient_get_handle(*clockComponent);
                       ^~~~~
player.c: In function ‘audioplay_get_latency’:
player.c:729:16: warning: variable ‘error’ set but not used [-Wunused-but-set-variable]
  OMX_ERRORTYPE error;
                ^~~~~
player.c: In function ‘receivepkt’:
player.c:823:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
player.c: In function ‘addnullpacket’:
player.c:908:16: warning: variable ‘padpacket’ set but not used [-Wunused-but-set-variable]
  unsigned char padpacket[2048];
                ^~~~~~~~~
player.c:904:6: warning: unused variable ‘err’ [-Wunused-variable]
  int err = 1000;
      ^~~
player.c: In function ‘main’:
player.c:1326:37: warning: passing argument 3 of ‘pthread_create’ from incompatible pointer type [-Wincompatible-pointer-types]
   if (pthread_create(&thread, NULL, receivepkt, oldnodecopy) != 0)
                                     ^~~~~~~~~~
In file included from player.c:6:
/usr/include/pthread.h:236:15: note: expected ‘void * (*)(void *)’ but argument is of type ‘void * (*)(Nodetype *)’ {aka ‘void * (*)(struct Node *)’}
       void *(*__start_routine) (void *),
       ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
player.c: In function ‘copy_into_buffer_and_empty’:
player.c:169:10: warning: ‘r’ may be used uninitialized in this function [-Wmaybe-uninitialized]
   return r;
          ^
cc -o player.bin -Wl,--whole-archive player.o -lilclient -lavformat -lavcodec -lavutil  -L/opt/vc/lib/ -lbrcmGLESv2 -lbrcmEGL -lopenmaxil -lbcm_host -lvcos -lvchiq_arm -lpthread -lrt -lm -L/opt/vc/src/hello_pi/libs/ilclient -L/opt/vc/src/hello_pi/libs/vgfont -Wl,--no-whole-archive -rdynamic
rm player.o
make[1]: Leaving directory '/home/idimou/lazycast/player'
make -C h264/.
make[1]: Entering directory '/home/idimou/lazycast/h264'
cc -DOMX_SKIP64BIT -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont   -DSTANDALONE -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -D_LINUX -fPIC -DPIC -D_REENTRANT -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -U_FORTIFY_SOURCE -Wall -g -DHAVE_LIBOPENMAX=2 -DOMX -DOMX_SKIP64BIT -ftree-vectorize -pipe -DUSE_EXTERNAL_OMX -DHAVE_LIBBCM_HOST -DUSE_EXTERNAL_LIBBCM_HOST -DUSE_VCHIQ_ARM -Wno-psabi -I/opt/vc/include/ -I /opt/vc/include/interface/vmcs_host/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux  -I/opt/vc/include/interface/vcos/  -I /opt/vc/include/IL  -I/opt/vc/src/hello_pi/libs/ilclient  -I/opt/vc/include/ -I/opt/vc/include/interface/vcos/pthreads -I/opt/vc/include/interface/vmcs_host/linux -I./ -I/opt/vc/src/hello_pi/libs/ilclient -I/opt/vc/src/hello_pi/libs/vgfont -g -c h264.c -o h264.o -Wno-deprecated-declarations
h264.c: In function ‘video_decode_test’:
h264.c:428:31: warning: passing argument 2 of ‘audioplay_create’ from incompatible pointer type [-Wincompatible-pointer-types]
  if (audioplay_create(client, &audio_render, list, 4) != 0)
[... 54 lines cut ...]
cc -o controlhidc.bin controlhidc.o -L /usr/X11R6/lib -lX11 -lm
cc -O2 -Wall  -g -c control.c -o control.o
control.c: In function ‘main’:
control.c:198:13: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 2 has type ‘KeySym’ {aka ‘long unsigned int’} [-Wformat=]
    printf("%x\n", *keysym);
            ~^     ~~~~~~~
            %lx
cc -o control.bin control.o -L /usr/X11R6/lib -lX11 -lm
rm control.o controlhidc.o
make[1]: Leaving directory '/home/idimou/lazycast/control'`

After this of course if I try to run all.sh I get a segmentation fault:
./all.sh: line 126:  4817 Segmentation fault      sudo wpa_cli -i$p2pdevinterface p2p_find type=progessive
./all.sh: line 126:  4827 Segmentation fault      sudo wpa_cli -i$p2pdevinterface set device_name "$(uname -n)"
./all.sh: line 126:  4836 Segmentation fault      sudo wpa_cli -i$p2pdevinterface set device_type 7-0050F204-1
./all.sh: line 126:  4845 Segmentation fault      sudo wpa_cli -i$p2pdevinterface set p2p_go_ht40 1
./all.sh: line 126:  4854 Segmentation fault      sudo wpa_cli -i$p2pdevinterface wfd_subelem_set 0 000600111c44012c
./all.sh: line 126:  4863 Segmentation fault      sudo wpa_cli -i$p2pdevinterface wfd_subelem_set 1 0006000000000000
./all.sh: line 126:  4872 Segmentation fault      sudo wpa_cli -i$p2pdevinterface wfd_subelem_set 6 000700000000000000

Same when I try to clone and use v1.0.1.

- Have you tested in on a pi zero? What can I do to make it?

- Should I try it on a pi 4?

homeworkc, 2020-09-30:
1. Is you pi zero the "W" version (i.e., with wifi)?
2. What is the system image you use?
3. I tested it on Pi 3 and 3+. It should also work on 4, zero W or any other versions with suitable external wifi card.
4. In order to make the project, the recommended method is to copy each command in the grey boxes one by one in README to the terminal. Be sure to hit enter after pasting commands or the last one won't execute. The release page only has outdated version and you should not use those (although it pre-compiles the codes so compilation is not required). It is recommended to run the commands locally and not over ssh as installing the NetworkManager will reset the (wired and wireless) network. If you must connect via ssh, skip the Install NetworkManager section.
5. From the message, it seems that the code actually compiles normally with a few noncritical warnings, and I don't think it's the compiled programs that causes the seg fault. It seems that ``wpa_cli`` is the one that actually causes the seg fault. Be sure to use up-to-date system image. You can also try using a separate SD card with the latest "Raspberry Pi OS (32-bit) with desktop Image with desktop based on Debian Buster" image.

idimou, 2020-09-30:
Thanks for the swift response.

1. yes pi zero W with wifi
2. Raspberry Pi OS (32-bit) Lite 20-08-2020 version
3. Good. I'd really like to make it work on the zero as it's cheaper and smaller.
4. I see. I run them one by one as you say. I didn't face any disconnection issues. Checking from my router admin page the pi zero received DHCP IPs for both WiFi and Ethernet. I'll try running the commands locally and let you know.
5. I used the latest Raspberry Pi OS (32-bit) Lite version (no desktop). If the lazycast needs the desktop version I'll install that.

idimou, 2020-09-30:
I've deployed it again on a the pi zero with Raspberry OS 20-08-2020 version with desktop (not Lite). It still gave me segmentation faults.
I tried the same on a pi 4 and it works. I see the "The display is ready" and rolling PIN messages. I was able to cast from an android phone using Castto app with 2 secs latency.

Using Windows 10 PC as a source the pi 4 appears on the available devices but each time I press on it the PIN textbox disappears, there is a drawer at the top of the win 10 screen indicating "Connecting to device" but nothing else happens. The console on the pi 4 side continues to show the PIN: .... scrolling.
- Is Miracast over Infrastructure (MICE) the only option to be able to use Win 10 as a source? Or should I be able to connect simply via p2p like I did form an android source?

homeworkc, 2020-09-30:
Both p2p and MICE should work with Win 10.

idimou, 2020-10-07:
hmmm still can't connect from windows 10 to a pi 4 running lazycast. Please test it with a pi 4 when you get a chance. Thanks again.

Istria1704, 2020-11-06:
Just for the info. I encountered the same error.
Using: 
old Raspberry Pi 1 B+
Fresh Raspbian Stretch Desktop from the README link (updated and upgraded)
Followed preparation steps in README without errors.

When running ./all.sh, I get the error.