homeworkc/lazycast #79: start lazycast from a script or on startup

patrickschaerer, 2021-02-09:
I d'like to start lazycast from a script (triggered through a dummy switch on domoticz home automation on a raspberry pi 3).
I installed lazycast an it works when I run it in CLI with ./all.h

I tried a this script lasycast_start.sh
`cd /home/pi/lazycast/
./all.sh`

After the execution of that script I can see the miracast device and connect to it. Unfortunately nothing is displayed on the hdmi out.

I tried to run lazycast when booting with Method2 SystemD. It was the same. I could see and connect to the miracast device but the mirrored screen was not shown on the hdmi.

Is that a bug or did I miss something?

joulester, 2021-03-19:
Did you solve this? Have same problem with systemd

patrickschaerer, 2021-03-21:
no couldn't solve it

iTommix, 2021-07-29:
Have the same issue. Anybody solved this?