homeworkc/lazycast #81: Work Issue

Leader-txt, 2021-03-06:
I have installed LazyCast on my RaspberryPi ZeroW,and my Android phone connected my pi successfully.
However, there was noting happened on my screen.
I used the commad--"./all.sh"to start lazycast.
After my phone connected,the command line show:
`./h264/h264.bin 41708 0 192.168.173.1 &
sh: 1: ./h264/h264.bin: not found`
What should I do to solve the problem?

LaurentG9661, 2021-03-08:
Hello, 
Kind of the same for me. I have installed Lazycast on a Raspberry Pi Zero WH, my source is an Android Samsung galaxy S20 FE 5G smartphone. I succeded to link correctly the smartphone to Raspberry with the WiFi direct but nothing more happened (with or without launch `./all.sh`) .

homeworkc, 2021-06-15:
"./h264/h264.bin: not found" means the binary is not compiled.
Please follow the instruction in https://github.com/homeworkc/lazycast#build-binaries. I suggest that you delete the current lazycast folder you have. Then copy every single command in the "build-binaries" paragraph to you terminal and run those commands.