homeworkc/lazycast #82: Sound Error with a Raspberry Pi Zero WH

LaurentG9661, 2021-03-17:
Raspberry Pi Zero WH
Raspbian GNU/Linux 10
lazycast version v1.2
release date: 2019-09-13
downloaded : March 2021

After launching `./all.sh` , when I connect my device (Galaxy S20 FE 5G) to my Raspberry Pi Zero WH, the connection is established succesfully and my display appears on the screen connected to the R.Pi .

```
p2p-dev-wlan0
wlan0
Selected interface 'p2p-wlan0-0'
Available interfaces:
p2p-wlan0-0
p2p-dev-wlan0
wlan0
already on
p2p-wlan0-0
The display is ready
Your device is called: raspberrypi
PIN:
31415926
PIN:
31415926
...
...
---- Negotiation successful ----
192.168.173.1
./h264/h264.bin 40610 0 192.168.173.1 &
idrport:40610
audiodest:0
sinkip:192.168.173.1
set audio backend :hdmi
oldcc 0 cc 14
idr:12
start:3169, end:3179
oldcc 4 cc 2
send idr
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
Content-Length: 17
Content-Type: text/parameters
CSeq: 103
...
```

But after a while, a "sound error" appears many times like in a row.

```
...
wfd_idr_request

idr:12
start:3434, end:3435
oldcc 5 cc 8
RTSP/1.0 200 OK
CSeq: 103
Session: 294702567;timeout=30

['RTSP/1.0 200 OK\r\nCSeq: 103\r\nSession: 294702567;timeout=30', '']
[]
send idr
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
Content-Length: 17
Content-Type: text/parameters
CSeq: 104

wfd_idr_request

RTSP/1.0 200 OK
CSeq: 104
Session: 294702567;timeout=30

['RTSP/1.0 200 OK\r\nCSeq: 104\r\nSession: 294702567;timeout=30', '']
[]
idr:12
start:47259, end:47263
oldcc 12 cc 15
send idr
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
Content-Length: 17
Content-Type: text/parameters
CSeq: 110

wfd_idr_request

RTSP/1.0 200 OK
CSeq: 110
Session: 294702567;timeout=30

['RTSP/1.0 200 OK\r\nCSeq: 110\r\nSession: 294702567;timeout=30', '']
[]
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
sound error
...
```

The problem seems to come from the h264.c file in h264/ which prints the "sound error" message, but I'm not sure...

I tried to switch the value of all the settings in `d2.py`, especially the sound_output_select and player_select. But the problem still exists.

Any idea how to fix this problem ?

Thanks for the help.

LaurentG9661, 2021-03-17:
And secondly, is it possible to turn on backchannel, to be able to use a keyboard connected to the usb port for instance ?

homeworkc, 2021-06-15:
1. "sound error" could be an error but it also could happen during normal operation when the source is sending sound samples faster than expected. You can ignore the message if the audio is working.
2. Backchannel is implemented and is enabled by default. However, I think some (or most) Android devices do not support.