homeworkc/lazycast #83: newer Windows versions disconnect after 2-3 secounds

KI20180151, 2021-04-15:
Hi, I installed your programm on my pi 3 b+ today.
On a laptop with an older WIndows version everything works fine but on an up-to-date version (I tried multiple devices) the feed works fine for 2-3 secounds but then it freezes and crashes.
I tried changing all kinds of setting and did troubleshooting for a few hours now but nothing helps.
When crashing it shows recv timeout.

homeworkc, 2021-06-15:
I will look into it but it may take a (long) while.
"recv timeout" means that the Raspberry Pi does not receive any wifi packet for an extended period of time. This usually indicates the problem with the wifi p2p connection. It could be that Windows change some logic of wifi p2p connections. (For example, switching channels.) It also could be that Windows adds some new wifi display features and somehow causes problems. (For example, switching to other video encoding schemes.)
One way to debug is using Wireshark both on the Windows side and on the Raspberry Pi side. A more comprehensive method to debug is to use a third device with its wifi card set to the monitor mode and with the Rx filter set to accept any frames. (E.g., another Raspberry Pi 3 with Kali Linux.)

trixrahbit, 2021-09-19:
I have the same issue on a pi4 with newer Windows. Will try to debug in the future if needed. Would love to see this working again.

Jaroslav-Dvorak, 2022-02-23:
I have two laptops. One of them works flawlessly. And the other disconnects after few seconds. I have no idea why. 
But I have noticed that in Wireshark is shown protocol of packets UDP on working laptop and MPEG TS or MPEG PES on disconnecting laptop.
I bought Microsoft Wireless Display Adapter and both laptops work. And in Wireshark is on both protocol UDP.
I spend a lot of time finding the problem - comparing source code from Miraclecast project with lazycast and so on..
Unfortunately I did not find the solution :-(

JpTiger, 2022-04-03:
Also having this issue with a new Windows 11 laptop. My old Win 10 laptop from 2014 works with Lazycast just fine. Happy to provide debugging info if helpful.

hj0707, 2022-04-09:
I have the same issue with raspberry Pi 3B + & windows10 21H1, even use network cable ,seems not wifi problem

Jhgallas, 2022-04-14:
Same, on 3B and Windows 11

Jhgallas, 2022-04-15:
> Same, on 3B and Windows 11

I have had success however with a different machine, with a 720p display (also on 11). However the resolution of my main pc does not appear to be what was causing problems - after changing the resolution of the lazycast display (painfully, in the 2-3 seconds of connection), as well as the resolution of my surface book, nothing changed, i get the same `recv timeout`

I did try to change to player 2 in the config file, wich immediately spits out some error codes:

`[rtp @ 0xb1a9d0] max delay reached. need to consume packet
[rtp @ 0xb1a9d0] RTP: missed 57 packets
[rtp @ 0xb1a9d0] PES packet size mismatch
[h264 @ 0xb1f2c0] sps_id 0 out of range
[h264 @ 0xb1f2c0] SPS unavailable in decode_picture_timing
[h264 @ 0xb1f2c0] non-existing PPS 0 referenced
[h264 @ 0xb1f2c0] sps_id 0 out of range
[h264 @ 0xb1f2c0] SPS unavailable in decode_picture_timing
[h264 @ 0xb1f2c0] non-existing PPS 0 referenced
[h264 @ 0xb1f2c0] decode_slice_header error
[h264 @ 0xb1f2c0] no frame!

[h264 @ 0xb1f2c0] error while decoding MB 55 29, bytestream -30
[h264 @ 0xb1f2c0] concealing 1609 DC, 1609 AC, 1609 MV errors in I frame

[rtp @ 0xb1a9d0] max delay reached. need to consume packet
[rtp @ 0xb1a9d0] RTP: missed 222 packets
[rtp @ 0xb1a9d0] PES packet size mismatch
`

Amusingly enough, however, after doing this it displays a completely frozen cast of my laptop screen until terminated

debritde, 2022-07-21:
Hi, i am also facing this problem with Windows 11.
Did anyone found a solution?

homeworkc, 2023-01-17:
Try using ``win10debug.sh`` instead of ``all.sh`` and see if it helps. (``win10debug.sh`` isn't very polished but it seems to at least keep the connection open.) Somehow newer Windows versions have some weird behaviors.

homeworkc, 2023-01-24:
Also please report the WiFi card model if you have connection issues. I've found a new Qualcomm card having difficulty to establish a connection in mixed 802.11ax environments. In such case, it might be beneficial to lock the wireless mode (on the Windows side) to 802.11n.