homeworkc/lazycast #90: Problems on Raspbian Buster

Xanth0k1d, 2021-08-22:
Hi,
On Buster, if I try to downgrade the wpasupplicant to 2.4, I'd get hit with the Device Not Ready problem in Network Manager.
Using wpa_cli yields a "segmentation fault".
I tried with Stretch/Buster without downgrading wpasupplicant, everything works, the device shows up on both windows and android, but I'get "Couldn't connect to the device". Even after entering the right password.
Do you have the version of raspbian which you last tested lazycast?

Xanth0k1d, 2021-08-23:
Hi, figured a few things out, here's a guide to those who are just trying to make things work:
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/2019-04-08-raspbian-stretch-lite.zip
^the raspian lite image I used, it's stretch. Was using lazycast1.1

//install the following packages (I was using lite as my sd card is only 4gb)
sudo apt update && sudo apt -y install 
net-tools python udhcpd //packages mentioned in README
network-manager network-manager-gnome openvpn openvpn-systemd-resolved network-manager-openvpn network-manager-openvpn-gnome //the network-manager stuff, you will need to reconnect to wifi again via network manager after you reboot
xserver-xorg raspberrypi-ui-mods lightdm //to get a basic desktop on raspbian lite.

//purge dhcpcd5
sudo apt -y purge dhcpcd5

//I did an upgrade just for the sake of it
sudo apt -y dist-upgrade

//disable systemd-resolved
sudo systemctl disable systemd-resolved

//Make sure to make things executable
cd lazycast1.1 //I had lazycast in this folder
chmod +x all.sh
chmod +x d2.py
cd player
chmod +x player.bin
cd ../h264
chmod +x h264.bin
cd ../control
chmod +x control.bin
chmod +x controlhidc.bin

//reboot just to clean things up a bit
sudo reboot

execute ./all.sh with root privilege to make sure things work.

JpTiger, 2022-02-19:
FWIW I tried doing this with a couple different versions of Stretch Desktop (specifically 2019-04-09 and 2019-11-13) on a Raspberry Pi Zero W and they don't to be compatible for some reason. Booting the OS for the first time always kicks me to rescue mode.