homeworkc/lazycast #91: Is it possible to disable PIN?

MilesFan, 2021-08-25:
This is a great project, and I am using it in a trusted environment that PIN is not necessary.

I know the Microsoft Wireless Display Adapter which also uses MiraCast technology can be set to not require PIN.

I know where to set a PIN in the source code, but not quite sure how to disable PIN.

`sudo wpa_cli -i$p2pinterface wps_pin any 31415926`

Does anyone have an idea?

Thanks!

homeworkc, 2021-09-07:
It might be possible but people have been trying this in the past and did not succeed. Basically the PIN is the mandatory connection method defined in WPS. WPS also has the optional push button method, which I think is what Microsoft adapter used. Right now the push button method is not implemented.

rtgiskard, 2023-12-09:
Tested with p2p mode, if set the property for `('fi.w1.wpa_supplicant1.Interface.WPS', 'ConfigMethods')` to `virtual_push_button`, then the sink will get `ProvisionDiscoveryPBCRequest` rather than `ProvisionDiscoveryRequestDisplayPin`. PBC does not requre the PIN code

phamthanhnam, 2023-12-21:
@rtgiskard How did you do that?

rtgiskard, 2024-01-17:
> @rtgiskard How did you do that?

I'm using the [dbus api](https://w1.fi/wpa_supplicant/devel/dbus.html#dbus_wps), just set `ConfigMethods` with the api on p2p setup (maybe right after `WfdIE`),