homeworkc/lazycast #92: latency issue

regie-techno, 2021-08-30:
I installed lazycast on a RPI4.
I cast a Windows 10 PC to RPI4 with 1920x1080 resolution
Lazycast works with VLC
It works really fine, but I have 5 seconds latency.
How could I reduce this latency ?

Thanks for your help

homeworkc, 2021-09-07:
VLC is not optimized for lowest latency. Using player 1 or player 2 is highly recommended since they are specifically written to reduce the latency. If you really want to use VLC, search "VLC latency" for methods to reduce the buffering at various stages.

regie-techno, 2021-09-08:
thanks for answering.

player1 or player2 do not work at home. The connexion is right, but the stream does not appear to my screen ... An idea ?
I am working on a RPI4 2Gb, with raspbian latest version.

purplepencil, 2024-05-20:
> thanks for answering.
> 
> player1 or player2 do not work at home. The connexion is right, but the stream does not appear to my screen ... An idea ? I am working on a RPI4 2Gb, with raspbian latest version.

Hi bro did you solve this problem?