homeworkc/lazycast #99: Missing Dependency libreadline7

myxxmikeyxx, 2021-11-08:
```
sudo apt --allow-downgrades install ./wpasupplicant_2.4-1+deb9u6_armhf.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'wpasupplicant' instead of './wpasupplicant_2.4-1+deb9u6_armhf.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 wpasupplicant : Depends: libreadline7 (>= 6.0) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

This was on the latest raspberry pi image (Bullseye I think). I tried to do `sudo apt-get install libreadline7 ` but it can not find the package. It has options for libreadline5 6 & 8 but not 7.

Oliver-Johnson, 2021-11-16:
> ```
> sudo apt --allow-downgrades install ./wpasupplicant_2.4-1+deb9u6_armhf.deb
> Reading package lists... Done
> Building dependency tree... Done
> Reading state information... Done
> Note, selecting [...]

I get exactly the same problem.

myxxmikeyxx, 2021-11-16:
I followed [this link](https://packages.ubuntu.com/bionic/amd64/libreadline7/download) to install it.
I think I still ran into other problems but if you do comment them and I will see what I can do to help.

ReggaeUlli, 2022-12-13:
I downloaded the file manually from a mirror listed on this site:
https://packages.debian.org/buster/libreadline7

For example:
`wget http://ftp.fr.debian.org/debian/pool/main/r/readline/libreadline7_7.0-3_armhf.deb`

And then installed it manually using:
 `sudo apt-get install ./libreadline7_7.0-3_armhf.deb`
 
 after that the downgrade of wpa_supplicant worked fine for me