homeworkc/lazycast #100: The latest release of Raspberry Pi OS comes with KMS as the default video driver

barhun, 2021-11-10:
Hi,

The latest release of Raspberry Pi OS is based on Debian Bullseye and comes with _KMS_ as the default video driver. With these new images, it seems not possible to build lazycast binaries since the C header files and libraries for Broadcom VideoCore APIs are no longer situated under the path `/opt/vc`. Is there any plan in progress to update the project to make it compatible with the new releases of the OS? You might want to use something like `gstreamer`, and AFAIK, this is how [RPiPlay](https://github.com/FD-/RPiPlay) supports video and audio decoding on other platforms lacking Raspberry Pi specific video APIs as well as for the 64-bit release of Raspberry Pi OS.

It is also worth mentioning that python2 is deprecated, discontinued and no more supported with Debian Bullseye and forth; however, as far as I've tested, this requires minimal changes to this project as the following:

- Replace the shebangs with `#!/usr/bin/env python` or `#!/usr/bin/env python3`
- Replace all the print statements in python scripts with print function calls

phamthanhnam, 2022-01-02:
On Raspberry Pi OS Bullseye, you can get back C headers and libraries in `/opt/vc` by cloning [userland](https://github.com/raspberrypi/userland) repository and running the script `buildme`.

For both lazycast and rpiplay, to use GPU players on Bullseye, you may need to disable DRM VC4 V3D driver (otherwise, HDMI output may be turned off right after receiving video stream), by editing `/boot/config.txt`:
- Comment out or delete the line `dtoverlay=vc4-kms-v3d`
- Add the line `dtoverlay=vc4-fkms-v3d` after the line [pi4]
- Reboot

If you don't want to use GPU players with deprecated Broadcom APIs, set `player_select = 0` in `lazycast/d2.py` to use vlc. There's also a boolean down in that file to use gstreamer instead (in function `launchplayer`).

closed 2022-03-14 by barhun

homeworkc, 2023-01-16:
Okay, thank you very much for finding this solution. I had also thought that Bullseye would be incompatible with lazycast. In that case I will modify the scripts and update the README.

RockNHawk, 2025-06-03:
> Okay, thank you very much for finding this solution. I had also thought that Bullseye would be incompatible with lazycast. In that case I will modify the scripts and update the README.

For arm64 OS, should add the option --aarch64 to the buildme command.

`./buildme --aarch64`