homeworkc/lazycast #101: Problems Running in Lite Raspbian Stretch

myxxmikeyxx, 2021-11-12:
I am using 2019-04-08-raspbian-stretch-lite.img Which i downloaded from [here](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/). This img file does have a newer WiFi package so I still downloaded the one on the README and installed it. I looked at issue #6 [here](https://github.com/homeworkc/lazycast/issues/6) but I can not get it to work. I got the install done and everything seems to run fine. I installed the `omxplayer` for testing and I also installed `libavformat-dev` but I can not get video or audio. I have the audio set to go out of the hdmi port. I have an HDMI to VGA that has a headphone jack and I have external speakers plugged in but no sound. I cant get video either but I am not even getting sound. 

Also sorry for all the post. I am just trying to get this project to work without needing a whole desktop. I have gotten it to work with a desktop environment before but I am not having much luck with the lite image.

barhun, 2021-11-13:
If you're sure you've done all the things mentioned in the README, what your setup lacks might be the required python packages: `dbus` and `gobject`. Try to run the following command and see if that helps:

sudo apt-get install -y python-dbus python-gobject

barhun, 2021-11-13:
Can you tell us how you start the script? Manually via an ssh session or automatically on boot through `/etc/rc.local`?

If it works when you start it via ssh, but not when it starts automatically; try attaching a PTS to the process via the `script` command.  Replace the command in `/etc/rc.local` with the following:

`script -q -c 'cd /path/to/lazycast; exec ./d2.sh' /dev/null >/dev/null 2>&1 &`

This was a problem I encountered back when I was running lazycast on Raspberry PI OS Lite.  Hope this helps.

myxxmikeyxx, 2021-11-13:
I am trying to run either way really but I’d like to try mice. I had already done the `apt python-dbus python-gobject` When I first saw those errors. I am trying to do it through terminal right now but will move it to rc.local after I get it to work. The problem is I can’t get it to work right now. I will try a new image this weekend hopefully and see if I can get it to work and then try the rc.local but the problem was I could not get it to display anything. I tried installing vlc and I tried installing omxplayer and doing the commands in issue #6 but no luck.