homeworkc/lazycast #102: Lazycast broken by bullseye's removal of python2

shodanx2, 2021-12-09:
Working on a fix, 
Managed to install python2 but not the dependant libraries, even with pip2 installed, dbus fails to install

barhun, 2022-01-01:
The step I followed to make it run on python3:

- Run the command `sudo apt-get install -y python3-dbus` in a terminal.
- Replace the statement `import gobject` in `mice.py` with `from gi.repository import GObject as gobject`
- Replace the shebangs with `#!/usr/bin/env python` or `#!/usr/bin/env python3`
- Replace all the print statements in python scripts with print function calls

JpTiger, 2022-02-18:
Any chance you (or someone else who's already done it ) would be willing to upload the modified mice.py to the repo with those changes?

barhun, 2022-03-16:
Opened the PR #106.

closed 2025-03-30 by homeworkc