homeworkc/lazycast #103: Minor visual glitch issue

shodanx2, 2022-01-03:
I've gotten lazycast to run pretty good, video quality is excellent, latency is low (although I have not measured it, I cannot easily observe any latency, I woud say I am certainly below 80ms maybe better)

However one issue that is always there is this periodic visual compression artifact.

See example video
https://www.youtube.com/watch?v=irpfqmFj_sY

I can only speculate but
I suspect this is happening on forced keyframes which may be of lower quality that the running stream, so every time there's one of those frame, the images glitches a little then corrects overtime with the in-between frames.

Maybe just increasing the delay between these frames, like once per 30 seconds ?  Or maybe only sending them when say, more than 30% of the pixels have changed ?

homeworkc, 2025-03-30:
I've looked into this issue quite thoroughly with my setup, but I haven't been able to pinpoint the exact cause. It seems somewhat related to the H.264 encoding on the sending machine. I have an AMD Ryzen laptop where I can reproduce the problem, but on an old Broadwell Intel laptop, the issue doesn't seem to occur. What's particularly strange is that if you capture the video stream (sent by the source) on the Pi's end, extract the TS stream, and play it back using a different media player like VLC, the glitches and breathing effects are still present.

Here are the steps to capture the video stream: First, install Wireshark on the Pi. After running all.sh and when the p2p interface is ready, use Wireshark to capture traffic on the p2p interface (use the p2p-wlan0-* interface, not p2p-dev-wlan0). Then, follow this guide (https://wiki.wireshark.org/mpeg_dump.lua) to convert the capture into a .ts file. You'll need to comment out the "debug("Dumped mpeg packets: " .. packet_count )" line to use it with recent versions of Wireshark. Finally, you can play the extracted .ts file using VLC or other media players.

At this point, I'm still unsure which part is causing the problem. It could be the H.264 encoding on the source machine, an issue with p2p networking, or something else entirely. If anyone has experience with H.264 debugging or the H.264 spec, I'd appreciate your input.

homeworkc, 2025-04-14:
This issue should be mitigated in the latest commit. I believe this issue is more related to how the source implements H.264 encoding. The glitches you see happen whenever there is a new IDR frame. Some sources' IDR frames are visually blurry compared to the immediately preceding frame, and IDR frames are sent periodically by default on some systems. In such cases, it is better to explicitly disable periodically sending IDR frames (and d2.py has been updated in the commit).

shodanx2, 2025-04-14:
Sorry, I could not help replicate my previous setup, it has long been superceded with sunshine/moonlight. 
The reason was that MS windows, would only allow to add a single screencast / Miracast over Infrastructure display
I don't know if this is still the case.