homeworkc/lazycast #104: Switching displays breaks functionality

JpTiger, 2022-03-10:
I used Lazycast successfully with a widescreen monitor on a Raspberry Pi Zero W. I used the exact same pi and configuration but unplugged the old monitor and plugged in an old 4:3 Dell Ultrasharp. The pi works fine with the display, and I can establish a remote connection with Lazycast, but I just get an Unnamed Window with a black screen upon connecting. The serving computer seems to think all is well, and will allow me theoretically to change resolutions, but Windows defaults to and recommends the resolution of the old monitor. This happens when trying to extend my display from two different Windows machines.

barhun, 2022-03-12:
Did you try deleting the file `edid.txt` located in the directory that you launch `lazycast` from? This is the file that provides possible configurations that your display supports and it gets generated only if there isn't one already in place, so you may need to delete it after connecting a different displays. (:

JpTiger, 2022-03-13:
> Did you try deleting the file `edid.txt` located in the directory that you launch `lazycast` from? This is the file which provides possible configurations that your display supports and it gets generated only if there isn't one already in place, so you may need to delete it after connecting a different display. (:

Makes logical sense but unfortunately doesn't work. Tried closing all.sh, deleting edid.txy and relaunching, but the issue persists.

JpTiger, 2022-03-13:
Odd development, there might be more going on here. I tried renaming the lazycast folder, cloning into a fresh copy of the github repo and compiling again. The issue _mostly_ still persists: that is Unnamed window opens, black, but then nothing else is displayed and everything else seems to think it's working properly. But this time when I go on the windows machine to see what resolution it's detecting, it's detecing the correct one.

I ran all.sh via SSH this time as an experiment, so I got the printout while it's running. Here it is:

```p2p-dev-wlan0
wlan0
Selected interface 'p2p-wlan0-0'
Available interfaces:
p2p-wlan0-0
p2p-dev-wlan0
wlan0
already on
p2p-wlan0-0
The display is ready
Your device is called: rpi0w
PIN:
31415926
PIN:
31415926
---M1--->
OPTIONS * RTSP/1.0
Require: org.wfa.wfd1.0
CSeq: 1

<--------
RTSP/1.0 200 OK
CSeq: 1
Public: org.wfa.wfd1.0, SET_PARAMETER, GET_PARAMETER

<---M2---
OPTIONS * RTSP/1.0
CSeq: 1
Require: org.wfa.wfd1.0

-------->
RTSP/1.0 200 OK
Server: MSMiracastSource/10.00.19041.1526 guid/73679708-34CC-0003-342D-7473CC34D801
Public: org.wfa.wfd1.0, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER
CSeq: 1

---M3--->
GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
Content-Length: 673
Content-Type: text/parameters
CSeq: 2

wfd_video_formats
wfd_audio_codecs
wfd_client_rtp_ports
wfd_display_edid
wfd_connector_type
wfd_uibc_capability
wfd2_rotation_capability
wfd2_video_formats
wfd2_audio_codecs
wfd2_video_stream_control
wfd_content_protection
wfd_idr_request_capability
intel_friendly_name
intel_sink_manufacturer_name
intel_sink_model_name
intel_sink_version
intel_sink_device_URL
microsoft_latency_management_capability
microsoft_format_change_capability
microsoft_diagnostics_capability
microsoft_cursor
microsoft_rtcp_capability
microsoft_video_formats
microsoft_max_bitrate
[... 167 lines cut ...]
Session: 93289466
Content-Type: text/parameters
CSeq: 5

wfd_trigger_method: TEARDOWN

PIN:
31415926
^CTraceback (most recent call last):
  File "./d2.py", line 58, in <module>
    sock.connect(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
KeyboardInterrupt
PIN:
31415926
^CTraceback (most recent call last):
  File "./d2.py", line 58, in <module>
    sock.connect(server_address)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
KeyboardInterrupt
PIN:
31415926
^[```

JpTiger, 2022-03-13:
Isolated the issue. It's not the new display, it's that I enabled the full KMS GL Driver for a different piece of software and forgot about it later. Disabling the driver and rebooting solved the issue. Closing this one and will open a new issue if that one hasn't already been reported.

closed 2022-03-13 by JpTiger