homeworkc/lazycast #110: Full on device crash on Raspberry Pi 4

Xyndra, 2022-07-08:
What I did:
Follow all instructions without getting Errors

What I can see:
The mouse sprung to the middle of the screen, the time stopped changing, the taskbar is shown and a window with the title "Unknown Window" whose borders are bigger than those of the Screen is in full-screen mode.

Xyndra, 2022-07-08:
Edit:
Now it does not crash the device anymore, however it still has the window whose borders only show on top and left side and it immediately kicks all devices as soon as they try to connect(Tried on two separate pcs).
Also it spams the pin which is kind of annoying

Xyndra, 2022-07-08:
Edit:
The immediate kick seems to come from the custom internet thing because it also denies any ssh access

Xyndra, 2022-07-08:
Un- and reinstalled gaining this knowledge:
The raspberry pi does not crash on first run, it just stops rendering until you restart via power cable.
Also it seems to help uninstall the vpn things. It is now able to maintain a connection for ~5 seconds

homeworkc, 2023-05-06:
This should be solved in the latest commit and the X window is no longer used.

closed 2023-05-06 by homeworkc