homeworkc/lazycast #113: P2P works but MICE does not

SIGO-26B, 2023-03-01:
Trying to get this running using MICE on a 3B+ running the lite version of Raspbian Buster pulled straight from raspberrypi.org using their imaging tool. I followed the steps to the letter in readme.md (except having to manually install git and python3-dbus as they weren't included in the image), and P2P cast seems to work just fine, but when I run mice.py, I get the following:

------------------------------------------------------------------------------------------------------------------------------
pi@rpi3:~/lazycast $ ./mice.py
dbus.Dictionary({dbus.String('P2PDeviceConfig'): dbus.Dictionary({dbus.String('DeviceName'): dbus.String('rpi3', variant_level=1), dbus.String('PrimaryDeviceType'): dbus.Array([dbus.Byte(0), dbus.Byte(7), dbus.Byte(0), dbus.Byte(80), dbus.Byte(242), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0)], signature=dbus.Signature('y'), variant_level=1), dbus.String('GOIntent'): dbus.UInt32(7, variant_level=1), dbus.String('PersistentReconnect'): dbus.Boolean(False, variant_level=1), dbus.String('ListenRegClass'): dbus.UInt32(0, variant_level=1), dbus.String('ListenChannel'): dbus.UInt32(0, variant_level=1), dbus.String('OperRegClass'): dbus.UInt32(0, variant_level=1), dbus.String('OperChannel'): dbus.UInt32(0, variant_level=1), dbus.String('IntraBss'): dbus.Boolean(True, variant_level=1), dbus.String('GroupIdle'): dbus.UInt32(0, variant_level=1), dbus.String('disassoc_low_ack'): dbus.UInt32(0, variant_level=1), dbus.String('NoGroupIface'): dbus.Boolean(False, variant_level=1), dbus.String('p2p_search_delay'): dbus.UInt32(500, variant_level=1)}, signature=dbus.Signature('sv'), variant_level=1), dbus.String('Peers'): dbus.Array([], signature=dbus.Signature('o'), variant_level=1), dbus.String('Role'): dbus.String('device', variant_level=1), dbus.String('Group'): dbus.ObjectPath('/', variant_level=1), dbus.String('PeerGO'): dbus.ObjectPath('/', variant_level=1), dbus.String('PersistentGroups'): dbus.Array([], signature=dbus.Signature('o'), variant_level=1)}, signature=dbus.Signature('sv'))

dbus.Dictionary({dbus.String('DebugLevel'): dbus.String('info', variant_level=1), dbus.String('DebugTimestamp'): dbus.Boolean(False, variant_level=1), dbus.String('DebugShowKeys'): dbus.Boolean(False, variant_level=1), dbus.String('Interfaces'): dbus.Array([dbus.ObjectPath('/fi/w1/wpa_supplicant1/Interfaces/1')], signature=dbus.Signature('o'), variant_level=1), dbus.String('EapMethods'): dbus.Array([dbus.String('MD5'), dbus.String('TLS'), dbus.String('MSCHAPV2'), dbus.String('PEAP'), dbus.String('TTLS'), dbus.String('GTC'), dbus.String('OTP'), dbus.String('SIM'), dbus.String('LEAP'), dbus.String('PSK'), dbus.String('AKA'), dbus.String("AKA'"), dbus.String('FAST'), dbus.String('PAX'), dbus.String('SAKE'), dbus.String('GPSK'), dbus.String('WSC'), dbus.String('IKEV2'), dbus.String('TNC'), dbus.String('PWD'), dbus.String('EKE')], signature=dbus.Signature('s'), variant_level=1), dbus.String('Capabilities'): dbus.Array([dbus.String('ap'), dbus.String('ibss-rsn'), dbus.String('p2p')], signature=dbus.Signature('s'), variant_level=1), dbus.String('WFDIEs'): dbus.Array([dbus.Byte(0), dbus.Byte(0), dbus.Byte(6), dbus.Byte(1), dbus.Byte(81), dbus.Byte(2), dbus.Byte(42), dbus.Byte(1), dbus.Byte(44), dbus.Byte(1), dbus.Byte(0), dbus.Byte(6), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(6), dbus.Byte(0), dbus.Byte(7), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0), dbus.Byte(0)], signature=dbus.Signature('y'), variant_level=1)}, signature=dbus.Signature('sv'))

./mice.py:187: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  gobject.MainLoop().get_context().iteration(True)
./mice.py:188: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  gobject.threads_init()
./mice.py:189: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  gobject.MainLoop().run()
groupStarted: dbus.Dictionary({dbus.String('interface_object'): dbus.ObjectPath('/fi/w1/wpa_supplicant1/Interfaces/5', variant_level=1), dbus.String('role'): dbus.String('GO', variant_level=1), dbus.String('group_object'): dbus.ObjectPath('/fi/w1/wpa_supplicant1/Interfaces/5/Groups/O9', variant_level=1)}, signature=dbus.Signature('sv'))
Group properties: dbus.Dictionary({dbus.String('Members'): dbus.Array([], signature=dbus.Signature('o'), variant_level=1), dbus.String('Group'): dbus.ObjectPath('/fi/w1/wpa_supplicant1/Interfaces/5/Groups/O9', variant_level=1), dbus.String('Role'): dbus.String('GO', variant_level=1), dbus.String('SSID'): dbus.ByteArray(b'DIRECT-O9', variant_level=1), dbus.String('BSSID'): dbus.ByteArray(b'\xf6\x9e\xe4\x9a(\xac', variant_level=1), dbus.String('Frequency'): dbus.UInt16(2412, variant_level=1), dbus.String('Passphrase'): dbus.String('AfuGMdak', variant_level=1), dbus.String('PSK'): dbus.ByteArray(b'c\xcd\xde\xfc\xf2\x8cO\x1d\xf6\xc7\xd3dn\x1d\x1fq\x10\xbba[%e@t\x16R\xf4\x86M\xa7\x80\xf3', variant_level=1), dbus.String('WPSVendorExtensions'): dbus.Array([], signature=dbus.Signature('ay'), variant_level=1)}, signature=dbus.Signature('sv'))
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "./mice.py", line 80, in groupStarted
    innerarray.append(dbus.Byte(c))
TypeError: Expected a bytes or str of length 1, or an int in the range 0-255
----------------------------------------------------------------------------------------------------------------

From there, it just hangs and I have to ctrl+c to stop it, then run removep2p.sh to be able to restart mice.py, which results in the above again. Can you help me figure out what's wrong? Thanks.

homeworkc, 2023-03-07:
The mice.py was written in python 2 with Buster image and haven't fully transitioned to python 3 yet. If you really want to try out MICE, I suggest using an older commit (before the python 3 transition, or perhaps using https://github.com/homeworkc/lazycast/tree/revert-106-master). You also might encounter some issues with the latest Windows. (I have only tested MICE with a (now) relatively old version of Windows 10.)

SIGO-26B, 2023-03-07:
Hey, thanks for the response. I'll try the revert process first and then the older commit, if need be. Whatever ends up working, I'll post details here for others.

SIGO-26B, 2023-03-08:
So, I went to revert to 106 and noticed that you updated all of the .py files and added a shell script for MICE. So, I wiped the ~/lazycast/ directory and re-cloned the repo. Everything is working as expected now and I do not see a WiFi Direct connection. Well done and thank you! I'll close out this issue.

closed 2023-03-08 by SIGO-26B