homeworkc/lazycast #114: Sending touch input to host

keesblok, 2023-03-28:
I am trying to send additional input events (touch screen events) from the Raspberry Pi to the host/source.

After looking at how the current events are handeled in `d2.py`, I would like to ask where I can find some documentation on what bytes to send to the host.

I know how to get the input from the touchscreen in python, just not what to send to the host.

So how do I know what to put in `m7` to get specific events working?

Thanks in advance!

keesblok, 2023-03-28:
After searching a bit more, I think I found it in the Miracast specification in the section "User Input Back Channel". If I have time and if can get it working, I might open a pull request to add the functionality to this repo

homeworkc, 2023-05-06:
FYI, there are two methods of sending inputs. One is "Generic" the other is "HIDC". Some older commits actually implemented both but the latest commits only use HIDC. The HIDC protocol is exactly the same in the USB spec and the traffic you'll see in USB captures, except that a few bytes are prepended to the front. For custom HIDC, you'll need to first send the HID descriptor (basically part of the USB descriptor during a normal USB enumeration) and afterwards, send the HID inputs ("m7" in the codes, but the "m7" in the context does not actually mean anything and it's just a random variable name).