homeworkc/lazycast #115: Loop on ./all about Selected interface

TiltedPhosphor, 2023-06-05:
I try to install it on a Raspberry Pi 3B+. And did every step, but when I start the all.sh File, it loops with 

`Selected interface 'p2p-dev-wlan0'
Available interfaces:
p2p-dev-wlan0
wlan0`

I looked in the code and it seems that it fails at the line 61
`result=$(sudo wpa_cli p2p_group_add -i$p2pdevinterface persistent$perstr $wlanfreq)`
I tried doing it manual outside of the script, but it also failed.

I tried looking in old issues of this and other projects which are related. But I cant seem to find find any solution.
If someone could help me, it would be very appreciated.

homeworkc, 2023-07-05:
Hi, what is the OS version on your Raspberry Pi 3B+? I used Raspberry Pi 3B+ as my main platform until recently so it should work. On older OS (and potentially on older Raspberry Pi), it might be preferable to install NetworkManager because the built-in WiFi management tools will mess with the creation of the p2p interface. (Installing NetworkManager might cause disconnection so you should use a spare SD card if you are not familar with the process.) The main purpose of all.sh is basically making sure that a p2p interface is set up properly.

privatedev11, 2023-07-14:
I had this issue, turns out it was an issue with ethernet, and it was fixed by connecting via WLAN. Are you using ethernet? If so connect via WLAN and let me know the results.

ThePrimeDev, 2023-10-22:
> Hi, what is the OS version on your Raspberry Pi 3B+? I used Raspberry Pi 3B+ as my main platform until recently so it should work. On older OS (and potentially on older Raspberry Pi), it might be [...]

I have same issue on raspberry Pi 3 Model B V1.2... Raspbian Bullseye

ThePrimeDev, 2023-10-23:
@homeworkc I installed networkmanager, enabled it, but same issue, altough that when I do nmcli device show then I get the wlan0 and p2p-dev-wlan0 both as disconnected(30) so the networkmanager sees them