homeworkc/lazycast #121: Supporting alternate display orientations

alexwhittemore, 2024-04-01:
I was able to compile and use lazycast per the readme with basically no fuss (after picking the right legacy RasPi OS version!), thanks!

My Pi is connected to a Waveshare 5.5" OLED touch screen that I'm trying to use as the target, but this display has a quirk that it adopts portrait orientation by default, and must be rotated in Linux for landscape use.

It seems Lazycast makes an assumption about the display orientation and just smacks the remote image center-screen rotated right-side-up, scaled to fit.

There's probably some easy spot to add in a 90* rotation to this, but I haven't looked yet.

[image]

homeworkc, 2024-04-02:
I believe this could be done but I haven't implemented it in h264.c yet. Omxplayer has an option to do rotation and from a quick look at its [code](https://github.com/popcornmix/omxplayer/blob/4037be549101f93082359f4f21c61c8c5f84a014/OMXVideo.cpp#L636), it seems that an extra parameter OMX_DISPLAY_ROT90 has to be set.

homeworkc, 2024-04-03:
Actually I found an easier method. Add this snippet at Line 422 of h264.c:
```
OMX_CONFIG_ROTATIONTYPE rotation;
memset(&rotation, 0, sizeof(rotation));
rotation.nSize = sizeof(rotation);
rotation.nVersion.nVersion = OMX_VERSION;
rotation.nRotation = 90;
rotation.nPortIndex = 90;
if (OMX_SetConfig(ILC_GET_HANDLE(video_render), OMX_IndexConfigCommonRotate, &rotation) != OMX_ErrorNone)
status = -15;
```

alexwhittemore, 2024-04-08:
I haven't had a chance to try that yet but thank you so much for suggesting it!