homeworkc/lazycast #124: Lazycast on Pi3 not compiling

kay102, 2024-05-22:
Hello all,

I don't get lazycast working on a Pi3 with bookworm 64 bit because [cmake ](https://github.com/homeworkc/lazycast#OS) fails. If I understand it correctly, my architecture is unknown.

I have attached the output of cmake.
[fehler.txt](https://github.com/homeworkc/lazycast/files/15405102/fehler.txt)

Is there anything I miss?

Kind regards

homeworkc, 2024-05-22:
Ideally you should run lazycast on "Raspberry Pi OS (Legacy, 32-bit), Bullseye". The cmake errors you see happened when compiling the userland library (https://github.com/raspberrypi/userland). You should probably look into if there are ways to compile the userland library in 64-bit environments. (Note that https://github.com/raspberrypi/userland says that 64-bit is not officially supported so it might not work at all.) But even if you get the 64-bit library working, I think Bookworm still has some issues so using 32-bit Bullseye is probably easier.

kay102, 2024-05-22:
Ok, thank you for the explanation!

closed 2024-05-22 by kay102