homeworkc/lazycast #125: PLEASE HELP!! The connection is made in player 1 and player 2, but my phone's screen is not reflected (the delay in VLC is very high in player 0)

purplepencil, 2024-07-12:
PLAYER 1:

---M1--->
OPTIONS * RTSP/1.0
CSeq: 1
Server: AllShareCast/Galaxy/3.1/NIBC
Require: org.wfa.wfd1.0

<--------
RTSP/1.0 200 OK
CSeq: 1
Public: org.wfa.wfd1.0, SET_PARAMETER, GET_PARAMETER

<---M2---
OPTIONS * RTSP/1.0
CSeq: 1
Require: org.wfa.wfd1.0

-------->
RTSP/1.0 200 OK
CSeq: 1
Public: org.wfa.wfd1.0, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

---M3--->
GET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 2
Content-Type: text/parameters
Content-Length: 200

wfd_video_formats
wfd_audio_codecs
wfd_uibc_capability
wfd_client_rtp_ports
wfd_content_protection
wfd_sec_screensharing
wfd_sec_portrait_display
wfd_sec_volume_support
wfd_sec_volume_status

Nothing written!
<--------
RTSP/1.0 200 OK
CSeq: 2
Content-Type: text/parameters
Content-Length: 460

wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play
wfd_audio_codecs: AAC 00000001 00
wfd_video_formats: 00 00 02 10 0001FEFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none
wfd_3d_video_formats: none
wfd_coupled_sink: none
wfd_connector_type: 05
wfd_uibc_capability: input_category_list=GENERIC, HIDC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=Keyboard/USB, Mouse/USB;port=none
wfd_standby_resume_capability: none
wfd_content_protection: none

---M4--->
SET_PARAMETER rtsp://localhost/wfd1.0 RTSP/1.0
CSeq: 3
Content-Type: text/parameters
Content-Length: 387

wfd_video_formats: 00 00 02 04 00000080 00000000 00000000 00 0000 0000 00 none none
wfd_audio_codecs: AAC 00000001 00
wfd_uibc_capability: input_category_list=GENERIC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=none;port=58506
wfd_uibc_setting: enable
wfd_presentation_URL: rtsp://192.168.173.80/wfd1.0/streamid=0 none
wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play

<--------
[... 438 lines cut ...]

<---M2---
OPTIONS * RTSP/1.0
CSeq: 1
Require: org.wfa.wfd1.0

-------->

---M3--->

Nothing written!
<--------
RTSP/1.0 200 OK
CSeq: 2
Content-Type: text/parameters
Content-Length: 461

wfd_client_rtp_ports: RTP/AVP/UDP;unicast 1028 0 mode=play
wfd_audio_codecs: LPCM 00000002 00
wfd_video_formats: 00 00 02 10 0001FEFF 3FFFFFFF 00000FFF 00 0000 0000 00 none none
wfd_3d_video_formats: none
wfd_coupled_sink: none
wfd_connector_type: 05
wfd_uibc_capability: input_category_list=GENERIC, HIDC;generic_cap_list=Keyboard, Mouse;hidc_cap_list=Keyboard/USB, Mouse/USB;port=none
wfd_standby_resume_capability: non

homeworkc, 2024-07-14:
Are you using Raspberry Pi 5 or 4 (or earlier)?

purplepencil, 2024-07-16:
sir, i am using raspberry pi 4